ros-drivers/pointgrey_camera_driver #32: ARM compatibilty

l0g1x, 2015-02-17:
Looked in the python script which downloaded the Flycapture SDK and there is no check for ARM (well actually it seems like it might be commented out). Would i be able to compile from source on arm7 (jetson tk1)? Would be nice to have since jetson is gigabit ethernet. Not sure if pointgrey's "flycapture.2.6.3.4_armhf" download is for arm7 or not though.

mikepurvis, 2015-02-17:
I believe it is. I would be delighted to merge a PR which detects ARM during the build and does the right thing. The intent of that script was always to provide for that possibility; I just never got around to implementing it.

l0g1x, 2015-02-17:
awesome. have a few exams this week so should get around to it on the weekend. thanks!

l0g1x, 2015-02-26:
What would be the best way to test whether everything installed correctly once i made changes?

mikepurvis, 2015-02-26:
There isn't a good way to test it in an automated way.

I'd be satisfied to simply verify that the package configures and builds correctly on i386, amd64, and armhf. In each case, after the build, run `catkin_make install`, and verify that the flycapture SO file is ending up in the lib directory.

l0g1x, 2015-03-02:
okay almost have it, the arm download from there website doesnt have any debs, its just all extracted already for you, so had to modify a bit, but catkin isnt linking the include files properly. Will finish this tomorrow.

l0g1x, 2015-03-02:
is the reason why the download_target script does "dpkg --extract" versus "dpkg -i", to prevent flycapture from updating to 2.7?

mikepurvis, 2015-03-02:
`dpkg -i` would require root to do an actual system install. I don't want a system install, I just want to extract the headers and so from the deb in order to place them in the build tree.

l0g1x, 2015-03-02:
the ARM download does not have any debs, once you extract the tar.gz, the folders (/include /lib /src /bin) already appear. The folder structure for the arm flycapture is quite different from the x86 download. 

For example: 
all the include files are right in the include folder (for ARM download) like so:
/include/FlyCapture2.h
versus the x86 download gives:
/include/flycapture/FlyCapture2.h

so then the compiler gives a error like so:

.....
In file included from /home/ubuntu/ros/igvc/src/pointgrey_camera_driver/pointgrey_camera_driver/src/PointGreyCamera.cpp:32:0:
/home/ubuntu/ros/igvc/src/pointgrey_camera_driver/pointgrey_camera_driver/include/pointgrey_camera_driver/PointGreyCamera.h:47:36: fatal error: flycapture/FlyCapture2.h: No such file or directory
 #include "flycapture/FlyCapture2.h"
                                    ^
compilation terminated.
----

Not sure how to go about this, but I can just simply use shutil.mkdir to create directory called "flycapture" and move all the header files in there.

mikepurvis, 2015-03-02:
Yeah, I would do that. Just use the Python script to perform whatever mangling is necessary to get it to work.

As part of your process, please check the 2.7.x flycapture downloads from Pt. Grey and confirm that nothing about the structure of the arm download has changed in the newer versions.

l0g1x, 2015-03-02:
550867d59c67820af38451ae53a11a74176f4c59

l0g1x, 2015-03-02:
2.7.x ARM structure seems the same, so we are good on that

mikepurvis, 2015-03-02:
A few comments added on the commit. Please open a PR, and we can continue the review there.

closed 2015-04-07 by l0g1x