ros-drivers/pointgrey_camera_driver #40: Not working on ubuntu 14.04

hordurk, 2015-04-10:
I'm having trouble using the driver on ubuntu 14.04 with a Grasshopper 3 USB3 camera. It was working fine on the same computer on 12.04. The nodelet runs but I don't see anything messages on the topics that get advertised. I get no errors.

Running list_cameras hangs for 20s before producing outputs. I checked out some of the flycap examples and they also seem to be having issues hanging, but at least they produced some output after an initial delay (not sure if the framerate of the output was correct though).

Is there anything you are aware that needs to be done in order for the driver / flycap to work on 14.04?

Thanks,

mikepurvis, 2015-04-10:
It worked for me with a Flea3 under 14.04, though I admit I haven't checked it recently.

hordurk, 2015-04-10:
Was that a USB3 model?

mikepurvis, 2015-04-10:
Yes it is.

hordurk, 2015-07-24:
No idea what changed, but it started working again on that same computer later on.

closed 2015-07-24 by hordurk