ros-drivers/pointgrey_camera_driver #43: Image wrong "shifted/cropped"

samuelba, 2015-06-19:
sometimes it happens that the image is cropped the wrong way, like the image below:

[image]

however, if I unsubscribe and resubscribe to the image topic, that can change. So after resubscribing the image can be cropped right or wrong, it's random.

I use Ubuntu 14.04 and the cameras are connected to USB 3.0

gnarang1024, 2016-08-03:
Hey, did you find a solution to this? I'm seeing the same behavior with my Chameleon 3 CM3-U3-13Y3C-CS sensor. Thanks!

samuelba, 2016-08-05:
After a driver update (don't remember which one) it worked without problems.

closed 2016-08-05 by samuelba