ros-drivers/pointgrey_camera_driver #45: Test driver with flycap SDK 2.8.3.1

mikepurvis, 2015-08-05:
We previously updated to 2.7.3.13 and then reverted to 2.6.3.4 due to #28. It would be great if someone could take on checking what the story is with 2.8.x and multiple cameras (as well as checking for other regressions).

mikepurvis, 2015-08-06:
I have verified that the driver compiles and works on bumblebee2 with 2.8.3.1, but have not checked multi-camera.

peterpolidoro, 2015-10-05:
I tried 2.8.3.1 with no other changes. It seems to work well with one camera. It allowed me to get a higher frame rate using a Pointgrey Grasshopper 3 than 2.6.3.4. When I tried two cameras, however, a Grasshopper 3 and a Flea 3, I could not get images from both cameras, only the one I connected to first.

konradb3, 2016-03-10:
I get the update on multi-camera problem from PTGrey support. They state that the problem have been fixed in 2.9.3.13 release. My initial testing with two Blackfly confirms that problem is gone.

laurimi, 2016-05-12:
The download link for 2.9.3.13 seems to no longer work on the Pointgrey website. The newest version 2.9.3.43 is available, but it seemed to be very slow for a single camera compared to 2.6.3.4. that currently works for a single camera in my application. However, neither 2.6.3.4. nor 2.9.3.43 work in the multi-camera case. This is for two Flea3 cameras, FL3-U3-13Y3M-C.