ros-drivers/pointgrey_camera_driver #50: Image Consistency Issues Exactly 50% of the time when rectifying

davelkan, 2015-10-23:
I am running ubuntu 14.04 with Indigo and a pointgrey blackfly. I'm booting a bunch of nodes as seen below.
```
<!-- Point Grey Blackfly -->

<arg name="camera_serial" default="0" />
<arg name="calibrated" default="0" />

  <group ns="camera">
    <node pkg="nodelet" type="nodelet" name="camera_nodelet_manager" args="manager" />
    <node pkg="nodelet" type="nodelet" name="camera_nodelet"
      args="load pointgrey_camera_driver/PointGreyCameraNodelet camera_nodelet_manager" >
      <param name="frame_id" value="$(arg sensor_frame_prefix)/camera_link" />
      <param name="serial" value="$(arg camera_serial)" />
      <param name="frame_rate" value="10" />
      <param name="camera_info_url" if="$(arg calibrated)"
        value="file://$(env HOME)/.ros/camera_info/$(arg camera_serial).yaml" />
    </node>
    <node pkg="nodelet" type="nodelet" name="image_proc_debayer"
      args="load image_proc/debayer camera_nodelet_manager">
    </node>
    <node pkg="nodelet" type="nodelet" name="rectify_mono"
      args="load image_proc/rectify camera_nodelet_manager">
    </node>
</group>

```
Every second time I run the launch file that contains this script, the camera does not work correctly and instead repeatedly prints the error posted below. When this occurs I do not get any images from the camera at all.  The rectify_mono node seems most linked to the issue as it mainly seems to occur when I'm trying to run that node.
```
[ERROR] [1445558727.357774357]: PointGreyCamera::grabImage Failed to retrieve buffer | FlyCapture2::ErrorType 41There is an image consistency issue with this image.
```
A common fix to this problem was to increase socket buffer size, but this does not seem to have helped.

Becojo, 2016-05-15:
@davelkan did you end up finding a solution to this problem? I'm experiencing exactly the same problem.

JaouadROS, 2017-04-05:
@Becojo I'm facing the same issue.. Did you solve it, please?
@davelkan

Becojo, 2017-04-05:
@JaouadROS Unfortunately, no. I accepted that the issue was consistent so whenever it fails, I just restart my nodes so it works correctly.