ros-drivers/pointgrey_camera_driver #52: Issues with running Bumbebee2 on Ubuntu 14.04 and ROS Indigo

StasiaS, 2015-12-08:
Hello!

I am installing pointgrey_camera_driver how writing in http://wiki.ros.org/pointgrey_camera_driver.
This two commands running succesful:
sudo apt-get install ros-indigo-pointgrey-camera-driver
rosrun pointgrey_camera_driver list_cameras

But following commands:
1) roslaunch pointgrey_camera_driver bumblebee.launch

[ERROR] [1449584576.466665790]: Reconfigure Callback failed with error: PointGreyCamera::setFormat7 Error validating Format 7 settings | FlyCapture2::ErrorType 22Error setting Format 7 information.
*** Error in `/opt/ros/indigo/lib/nodelet/nodelet': double free or corruption (out): 0xaf368510 ***
[FATAL] [1449584581.254265545]: Service call failed!
[camera/bumblebee_nodelet_manager-1] process has died [pid 22866, exit code -6, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=bumblebee_nodelet_manager __log:=/home/anastasia/.ros/log/1e99c942-9db7-11e5-9030-f079595ab174/camera-bumblebee_nodelet_manager-1.log].
log file: /home/anastasia/.ros/log/1e99c942-9db7-11e5-9030-f079595ab174/camera-bumblebee_nodelet_manager-1*.log
[camera/image_proc_debayer_right-4] process has died [pid 22869, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load image_proc/debayer bumblebee_nodelet_manager image_raw:=right/image_raw image_mono:=right/image_mono image_color:=right/image_color __name:=image_proc_debayer_right __log:=/home/anastasia/.ros/log/1e99c942-9db7-11e5-9030-f079595ab174/camera-image_proc_debayer_right-4.log].
log file: /home/anastasia/.ros/log/1e99c942-9db7-11e5-9030-f079595ab174/camera-image_proc_debayer_right-4*.log

2) roslaunch pointgrey_camera_driver bumblebee.launch
[ERROR] [1449584679.303730179]: Cannot load nodelet /camera/camera_nodelet for one exists with that name already
[ERROR] [1449584679.311160908]: Cannot load nodelet /camera/image_proc_debayer for one exists with that name already
[FATAL] [1449584679.359448473]: Service call failed!
[FATAL] [1449584679.367674045]: Service call failed!

3) roslaunch pointgrey_camera_driver camera.launch camera_serial:=15206900

[FATAL] [1449584679.303966112]: Service call failed!
[FATAL] [1449584679.311365468]: Service call failed!
[ERROR] [1449584679.358053649]: Failed to find nodelet with name '/camera/camera_nodelet' to unload.
[ERROR] [1449584679.366798929]: Failed to find nodelet with name '/camera/image_proc_debayer' to unload.
[camera/camera_nodelet-2] process has died [pid 24110, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load pointgrey_camera_driver/PointGreyCameraNodelet camera_nodelet_manager __name:=camera_nodelet __log:=/home/anastasia/.ros/log/1e99c942-9db7-11e5-9030-f079595ab174/camera-camera_nodelet-2.log].
log file: /home/anastasia/.ros/log/1e99c942-9db7-11e5-9030-f079595ab174/camera-camera_nodelet-2*.log
[camera/image_proc_debayer-3] process has died [pid 24111, exit code 255, cmd /opt/ros/indigo/lib/nodelet/nodelet load image_proc/debayer camera_nodelet_manager __name:=image_proc_debayer __log:=/home/anastasia/.ros/log/1e99c942-9db7-11e5-9030-f079595ab174/camera-image_proc_debayer-3.log].
log file: /home/anastasia/.ros/log/1e99c942-9db7-11e5-9030-f079595ab174/camera-image_proc_debayer-3*.log

What may be associated with these errors?

phuicy, 2016-06-03:
Any answers on this one?

HyunJun-Jo, 2016-12-26:
I have a similar error.

When I try to run the 'roslaunch pointgrey_camera_driver bumblebee.launch' command, It looks like going well below. 

-------------------------------------------------------------------------------------------------------------------
user@NRMK-STEPi7:~$ roslaunch pointgrey_camera_driver bumblebee.launch
... logging to /home/user/.ros/log/1f740a84-cb39-11e6-bc6d-d050992f3cb7/roslaunch-NRMK-STEPi7-4573.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://127.0.0.1:60645/

SUMMARY
========

PARAMETERS
 * /camera/bumblebee_nodelet/first_namespace: left
 * /camera/bumblebee_nodelet/format7_color_coding: raw16
 * /camera/bumblebee_nodelet/frame_id: bumblebee
 * /camera/bumblebee_nodelet/frame_rate: 15
 * /camera/bumblebee_nodelet/second_namespace: right
 * /camera/bumblebee_nodelet/serial: 0
 * /rosdistro: indigo
 * /rosversion: 1.11.20

NODES
  /camera/
    bumblebee_nodelet (nodelet/nodelet)
    bumblebee_nodelet_manager (nodelet/nodelet)
    image_proc_debayer_left (nodelet/nodelet)
    image_proc_debayer_right (nodelet/nodelet)

ROS_MASTER_URI=http://127.0.0.1:11311

core service [/rosout] found
process[camera/bumblebee_nodelet_manager-1]: started with pid [4591]
process[camera/bumblebee_nodelet-2]: started with pid [4592]
process[camera/image_proc_debayer_left-3]: started with pid [4605]
process[camera/image_proc_debayer_right-4]: started with pid [4606]

-------------------------------------------------------------------------------------------------------------------

But after few seconds, I have error below. 

-------------------------------------------------------------------------------------------------------------------
[ERROR] [1482735701.847689953]: PointGreyCamera::connect could not set GigE packet_size | FlyCapture2::ErrorType 29 Failed to write SCPS0 register
-------------------------------------------------------------------------------------------------------------------

Is there any solution?