ros-drivers/pointgrey_camera_driver #55: Green Image

zfxw0206, 2016-01-20:
Now I am using Ubuntu 14.04.3 and  ROS indigo to run blackfly GigE. After running "roslaunch camera.launch" and rviz, I could get the picture from the camera, but the problem is the picture is green.

I think it is white balance problem. Does anybody know how to fix it?

Thank you!

zfxw0206, 2016-01-22:
[image]

mikepurvis, 2016-01-22:
Looks like a [bayer problem](https://en.wikipedia.org/wiki/Bayer_filter). PGR cameras are in theory supposed to send their own bayer information, so that processing software can effectively recover the colour, but in practice this seems to work only some of the time:

https://github.com/ros-drivers/pointgrey_camera_driver/blob/master/pointgrey_camera_driver/src/PointGreyCamera.cpp#L944

We should probably just put in a param that lets you override what the bayer setting in the Image msg is, to give users an escape hatch when this happens.

zfxw0206, 2016-01-22:
Do you mean change line 944 from "std::string imageEncodeing = sensor_msgs::image_encodings::MONO8;" to "std::string imageEncodeing;"?

I tried it, it doesn't work.

When I run the SDK "FlyCap2", I still get the green image,after opening the "Configuration", I found the White Balance(red) and White Balance(blue)  is zero. Even though I change it to "auto" to adjust it to normal image, when I run launch file next time, the image is still green.

So I am thinking is there a configuration file to adjust the white balance.

mikepurvis, 2016-01-22:
I could be wrong, but this definitely looks like a decoding problem more than a white balance problem.

However, if even the official Point Grey tools are giving you trouble, then it's possible your camera is malfunctioning. Most of my experience here is with the Firewire cams, so I don't know what the story is with GigE ones, but are you able to view the stream in a tool like Coriander? Coriander is especially nice because it lets you override the bayer setting right in its GUI, so you can quickly find out what the correct setting is.

mikepurvis, 2016-01-22:
The fix I'm envisioning (at least for you, in the short term) would be to eliminate that whole if/switch-block and replace it with a simple override, like:

    imageEncoding = sensor_msgs::image_encodings::BAYER_RGGB8;  # This is the right bayer config for my camera

zfxw0206, 2016-01-22:
I found another driver here: 
https://github.com/beta-robots/btr-ros-pkg

With this driver, I could get the normal image by running rosla

mikepurvis, 2016-01-22:
Interesting, looks like that one uses three-byte pixels:

https://github.com/beta-robots/btr-ros-pkg/blob/0caaae56c70094d38ba13c676615bf91c413da41/btr_bfly_camera/src/btr_bfly_camera_node.cpp#L141

Anyhow, glad you got it resolved to your satisfaction.

closed 2016-01-22 by mikepurvis

xuqingwenkk, 2016-12-01:
Using rqt_reconfigure can adjust white balance but cannot store it