ros-drivers/pointgrey_camera_driver #58: Bandwith and framerate issues

gscortichini, 2016-02-25:
Hi everyone, i`m using a Pointgrey Blackfly Camera (BFLY-PGE-23S6C, 1920x1200)  through ROS pointgrey_camera_driver.
My camera works perfectly on FlyCap2, where i can easly reach 40Hz of framerate (the max framerate for that camera model is 41).

When i work with point_grey_camera driver with this launch file:
-----------------------------------------------------------------------------------------------------------------
<launch>
   
  <arg name="camera_serial" default="0" />
  <arg name="calibrated" default="0" />

  <group ns="camera">
    <node pkg="nodelet" type="nodelet" name="camera_nodelet_manager" args="manager" />

    <node pkg="nodelet" type="nodelet" name="camera_nodelet"
          args="load pointgrey_camera_driver/PointGreyCameraNodelet camera_nodelet_manager" >
      <param name="frame_id" value="camera" />
      <param name="serial" value="$(arg camera_serial)" />

      
      	   <param name="frame_rate" value="35" />
      	   <param name="format7_color_coding" value="Mono8" /> 
      
      <param name="camera_info_url" if="$(arg calibrated)"
             value="file://$(env HOME)/.ros/camera_info/$(arg camera_serial).yaml" />
    </node>

    <node pkg="nodelet" type="nodelet" name="image_proc_debayer"
          args="load image_proc/debayer camera_nodelet_manager">

    </node>
  </group>
</launch>

----------------------------------------------------------------------------------------------------------
I can reach only 12Hz of framerate on "camera/image_raw" topic. 
It doesn`t matter which videomode i choose, the max framerate is always 12Hz.

I tried Raw8,Raw16,Mono8 and Mono16. I get always the same max framerate.
This means that the Bandwith is variable, but not directly by setting one specific parameter.

After this, when i run Flycap2, i noticed that the Packet Size and Packet Delay values were changed, and obviously the Bandwith too.

Is there anything i can do in order to increase the framerate and reach 35Hz or 40Hz?

cynosure4sure, 2016-04-21:
did you find any solution for this? I am using Chameleon 3 on one of my older computer it works fine but on the new one I can't get a frame rate higher than 5.6

peterpolidoro, 2017-08-10:
I also get a much higher frame rate when I use FlyCap2 than I do using this driver. The maximum frame rate I see on the camera/image_raw topic is a little more than half of the maximum frame rate I get using FlyCap2. No settings that I can change using rqt_reconfigure seem to help this. I do not know if this is due to some overhead of wrapping the PointGrey driver with the ROS driver or if I am just not setting some values properly.

I am using a GS3-U3-23S6M camera with a Jetson TX1. The maximum camera frame rate is supposed to be about 163 at 1900x1200. Using FlyCap2 I can get it up to about 100 by playing with various settings. When I run this driver, the maximum frame rate on the camera/image_raw topic is about 60 Hz.

I have noticed that the maximum frame rate increases in FlyCap2 when I switch to Raw8 from Mono8 and when I use the largest packet size. I do not seem to be able to switch to Raw8 using this driver and I do not know how to change the packet size. I am not sure what else to try to improve the image topic frame rate.

JaouadROS, 2017-10-25:
@brianabrams It works but you need to launch **rqt_reconfigure** everytime you use the driver?

CaptainSpuerSlow, 2019-03-18:
Try turn off auto_shutter and manual set the shutter speed in the launch file.