ros-drivers/pointgrey_camera_driver #61: Errors after power on the camera

zfxw0206, 2016-03-23:
I am using Ubuntu 14.04 and ROS indigo to run two pointgrey GigE blackfly cameras.
Now the problems is every time after I power on the cameras and run the "camera.launch" file, I always get the following error:

[ERROR] [1458741761.509388400]: PointGreyCamera::start Failed to start capture | FlyCapture2::ErrorType 22Could not get the current video mode and frame rate.

[ERROR] [1458741936.227051214]: PointGreyCamera::connect could not discover GigE packet_size | FlyCapture2::ErrorType 1Failed to discover GigE packet size

[ERROR] [1458741969.818965453]: PointGreyCamera::start Failed to start capture | FlyCapture2::ErrorType 33Error starting isochronous stream.

But when I tried to detect the cameras with rostopic "/diagnostics", I got nothing. I have to restart the launch, then it works fine. Does anybody could help me to fix these problems or tell me how to detect these errors? Thank you!

fabrizioschiano, 2016-12-12:
Could you show us your launch file? 

What happens if you do :
`rosrun pointgrey_camera_driver list_cameras `

zfxw0206, 2017-08-22:
My problem is still there, here is my camera information:
[0]Serial: 16162102, Model: Blackfly BFLY-PGE-23S2C, Vendor: Point Grey Research, Sensor: Sony IMX136 (1/2.8" Color CMOS), Resolution: 1920x1200, Color: true, Firmware Version: 1.44.3.0

bnjff, 2017-08-30:
`sudo sh -c 'echo 1024 > /sys/module/usbcore/parameters/usbfs_memory_mb'`

zfxw0206, 2017-08-30:
My camera is GigE camera, why set it to usbcore?

bnjff, 2017-08-30:
Ah, sorry, I didn't notice it was GigE. Increasing usbfs memory limit solved "starting isochronous stream" error with USB3 camera.

sheelabhadra, 2017-10-27:
Hi @zfxw0206 , were you able to solve the issue? I am facing a similar problem with a BlackFly BFLY-PGE-20E4C camera.

sheelabhadra, 2017-10-27:
I didn't get the error anymore after I closed the FlyCapture application and ran the launch file in the terminal. I am using ROS Kinect by the way.

royaalto, 2017-11-27:
I got the same problem when run 'roslaunch pointgrey_camera_driver bumblebee.launch'

[ERROR] [1511783092.590248274]: Reconfigure Callback failed with error: PointGreyCamera::setFormat7 Format 7 mode not supported on this camera.
[ERROR] [1511783092.922220934]: PointGreyCamera::start Failed to start capture | FlyCapture2::ErrorType 33 Error starting isochronous stream.

Number of cameras found: 2
[0]Serial: 15152092, Model: Firefly MV FMVU-03MTM, Vendor: Point Grey Research, Sensor: Micron MT9V022177ATM (1/3" 640x480 CMOS), Resolution: 752x480, Color: false, Firmware Version: 1.8.3.0
[1]Serial: 14381259, Model: Firefly MV FMVU-03MTM, Vendor: Point Grey Research, Sensor: Micron MT9V022177ATM (1/3" 640x480 CMOS), Resolution: 752x480, Color: false, Firmware Version: 1.8.3.0

can anyone help?

zfxw0206, 2017-12-06:
@sheelabhadra , I still have this issue, what I did is restarting the driver after getting the error, then the error disappeared.

SubMishMar, 2018-03-27:
@sheelabhadra now I have this issue, can you please help me with this if you solved it somehow?

fabrizioschiano, 2018-03-27:
@SubMishMar did you look at this : https://github.com/ros-drivers/pointgrey_camera_driver/commit/db22db0c7e08230066f37f2cbd37e6012476eede

Are you sure to be running the last version of the driver? (did you do a `git pull` and a `catkin_make`)

SubMishMar, 2018-03-28:
@fabrizioschiano I installed using sudo apt-get install, I dont think i need to git pull.. do I?

fabrizioschiano, 2018-03-29:
@SubMishMar I am not sure, I don't think you need to do a git pull but just verify you have the following: https://github.com/ros-drivers/pointgrey_camera_driver/commit/db22db0c7e08230066f37f2cbd37e6012476eede

Maybe, the problem is somewhere else.