ros-drivers/pointgrey_camera_driver #62: Problem with Pointgrey USB3 camera

zachfang, 2016-03-31:
I am using Ubuntu 14.04 with a USB-3 Pointgrey camera. Everything looks no problem that I can find the camera and receive correct information using list_cameras launch file. When I run camera.launch, it seems working properly as well that bunch of topics will be published. However, I could't get image from any of the topics by image_view or RVIZ. Them just show a gray image without any error. 

Some output listed below:
rosrun image_view image_view image:=/camera/image_raw
[ INFO] [1459439160.156610663]: Using transport "raw"

Any suggestion or help will be greatly appreciated. Thanks.

jhoare, 2016-04-11:
Did you increase the usbfs size as per: https://www.ptgrey.com/tan/10685 ?

zachfang, 2016-04-28:
Yes, thanks. I have solved the problem! Setting larger memory and put raw1394 module in the file seems to be the reason I failed before.

closed 2016-04-28 by zachfang