ros-drivers/pointgrey_camera_driver #63: USB3 Point Grey Chameleon3 CM3-U3-13S2C (colour) outputs no colour images (only grey).

Crusty82, 2016-04-12:
Model: Chameleon3 CM3-U3-13S2C
ROS: Indigo
Machine: 64 Bit Intel
OS: 64Bit Ubuntu 14.04.1

Further context here: http://answers.ros.org/question/231635/ros-point-grey-chameleon-usb3-cam-no-colour-images/

Temporarily resolved by switching to Kumar Robotics' ROS stack for Point Grey USB3 Cameras, which works for me.

m91moreira, 2016-06-14:
Hi,

I'm having the same problem with the same camera model.

I tryed to use the Kumar Robotics stack, but I'm only abble to get no color images. I've tryed all the video modes without success...

What video mode did you choose?

Crusty82, 2016-06-14:
Video Mode 23.

m91moreira, 2016-06-14:
I've tried before and it didn't work.

Did you change your launch file?

Crusty82, 2016-06-14:
No, I didn't change the launch file. What topic are you looking at when visualizing the image?

m91moreira, 2016-06-14:
After running the launch file, I run the image_proc, in order to get the color images, but I get a warning saying " Color topic '/pg_15374753/image_color' requested, but raw image data from topic '/pg_15374753/image_raw' is grayscale".

I'm trying to look to /image_color topic.

m91moreira, 2016-06-27:
It seems that the problem is in the flycapture lib.

It doesn't get the isColorCamera field correctly on armhf. In order to get color on the odroid, I forced all the verifications of isColorCamera to true.

jgurzoni, 2016-10-27:
This issue seem to have been resolved with the library compilation against the 2.9.3.43 library

wynn4, 2017-05-25:
I'm having the same problem using Chameleon3, Ubuntu 16.04 LTS, Odroid XU4, and flycapture.2.11.3.121_armhf.  The only place I can see to change the isColorCamera flag is in flycapture.2.11.3.121_armhf/include/FlyCapture2Defs.h.  Setting it to true and I'm still only getting a mono8 grayscale image. Am I missing something?  rosrun pointgrey_camera_driver list_cameras still says that Color is false

IanBoyanZhang, 2017-08-24:
Hi @wynn4, were you able to fix the issue? We are having similar issues with "Point Grey Grasshopper 3". Thanks!

wynn4, 2017-08-24:
@IanBoyanZhang, are you also using an ARM processor such as the Odroid XU4 or TX1?  Have you installed it via apt-get, or have you built it from source.  In our case, the color image didn't work when using apt-get, but it did work when we built it from source.  You may also have to grep for 'isColorCamera' in the flycapture lib and force it to be true.  When building from source, there is a header file that won't be found (flycapture2.h I think) but all you have to do is add the location of that file as an include directory in the CMakeLists.txt.  Hope that helps.

IanBoyanZhang, 2017-08-27:
@wynn4 Thanks for getting back to me! We are running it on PC instead of an ARM SoC. We fixed the CMakeLists. It resolves the problem. Thanks again.