ros-drivers/pointgrey_camera_driver #77: [Request] release pointgrey_camera_driver on armhf

furushchev, 2016-08-17:
I use NVIDIA Jetson TX1 which has armhf 32bit architecture, and it seems that only `pointgrey_description` is released and `pointgrey_camera_driver` is not yet released.

```bash
ubuntu@tegra-ubuntu:~$ apt-cache search pointgrey
ros-indigo-pointgrey-camera-description - A series of URDF driver decriptions for Point Grey cameras
ros-jade-pointgrey-camera-description - A series of URDF driver decriptions for Point Grey cameras
```

It is very useful to release this driver package as apt package, because building on arm machine takes a lot time.

furushchev, 2017-08-11:
Closed due to inactivity.

closed 2017-08-11 by furushchev

mikepurvis, 2017-08-14:
armv8 is supported, so you should be good for the TX1, at least after the next release (which will include ef74de62985baa4ad83e7add38968b10701c8538).