ros-drivers/pointgrey_camera_driver #83: Not able to set the video_mode "format7_mode4" for a flea FL3-U3-32S2C

fabrizioschiano, 2016-11-08:
Hi,

I am trying to set a different mode for my flea FL3-U3-32S2C. I can do through the flycapture GUI but not through a launch file. In particular I would like to set: 
`      <param name="video_mode" value="format7_mode4" />`

The launch file I am using is the following. 
```
<launch>
   <!-- Determine this using rosrun pointgrey_camera_driver list_cameras.
       If not specified, defaults to first camera found. -->
  <arg name="camera_serial" default="15357017" />
  <arg name="calibrated" default="1" />

<!-- The camera flea3 that we have has the following Serial number : 15357017 -->

  <group ns="camera">
    <node pkg="nodelet" type="nodelet" name="camera_nodelet_manager" args="manager" />

    <node pkg="nodelet" type="nodelet" name="camera_nodelet"
          args="load pointgrey_camera_driver/PointGreyCameraNodelet camera_nodelet_manager" >
      <param name="frame_id" value="camera" />
      <param name="serial" value="$(arg camera_serial)" />

      <!-- When unspecified, the driver will use the default framerate as given by the
           camera itself. Use this parameter to override that value for cameras capable of
           other framerates. -->
      <param name="frame_rate" value="25" />
      <param name="video_mode" value="format7_mode4" />
      <!-- Use the camera_calibration package to create this file -->
      <param name="camera_info_url" if="$(arg calibrated)"
             value="file://$(env HOME)/.ros/camera_info/$(arg camera_serial).yaml" />
    </node>

  </group>
</launch>

```

The goal would be to change the resolution of the image without changing the Region of Interest. I know it should be possible but until now I did not succeed to do it via ROS. 

Thanks for any help,
Fabrizio

fabrizioschiano, 2016-11-08:
I added to the file PointGreyCamera.cpp 

```
  else if(vmode.compare("format7_mode4") == 0)
  {
    fmt7Mode = MODE_4;
    vmode_out = VIDEOMODE_FORMAT7;
  }
```
and it is working. 

Is that a good solution?

Thanks.

fabrizioschiano, 2016-11-09:
You can see my entire modification [here](https://github.com/fabrizioschiano/pointgrey_camera_driver/commit/35d9cf4d7d6a64e559da1430907a498e4a702c5c)

Let me know if it makes sense.

mikepurvis, 2016-11-09:
Looks reasonable, please send a PR. Thanks!

fabrizioschiano, 2016-11-09:
Done.

commit db22db0c (2016-11-09): Added the option to use the format7_mode4. (#84)

This allows to use the mode4 which corresponds to a 2x2 binning.

Example: an image of 1000x800 becomes an image of 500x400.

I tested on a flea usb 3.0 camera.

Closes #83.
closed 2016-11-09 by mikepurvis