ros-drivers/pointgrey_camera_driver #85: This package does not automatically install 60-pgr.rules

TheDash, 2016-11-09:
Manual install and copy/paste required before camera is detected.

fabrizioschiano, 2016-11-09:
Could you describe a bit more your problem ? I did not understand what is it about.

TheDash, 2016-11-09:
The udev rules aren't installed by this package. So when you install the camera, the package won't work unless you configure the udev rules. There is source code in the package that means to do this - but it doesn't achieve it.

mikepurvis, 2016-11-09:
It does if you install from the deb— debhelper picks up the files and slips them into the correct locations:

https://github.com/ros-drivers-gbp/pointgrey_camera_driver-release/tree/debian/indigo/pointgrey_camera_driver/debian

TheDash, 2016-11-09:
It was installed by the deb. Version 0.12.2. Files were put into /lib/udev/rules.d but not /etc/udev/rules.d

After they were applied to /etc/udev/rules.d, it detected the camera

TheDash, 2016-11-09:
perhaps its related to the files being there - but a lack of a command to actually make sure these commands are installed.

TheDash, 2016-11-09:
I'll close this. seems like it did install. But for some reason it wasn't showing up and I had to manually do it. I re-installed later to see if it was working as intended, and it did.

Must have been a blip in the matrix.. We got cheesed.

closed 2016-11-09 by TheDash