ros-drivers/pointgrey_camera_driver #94: Frame Rate Issue

bobjohnson2040, 2017-02-07:
I have a Point Grey Flea3 camera using pointgrey_camera_driver. Every time I launch it in ROS the frame rate is 7 FPS. I am able to change the frame rate to 30 FPS using rqt_reconfigure. I would like for the camera to start at 30 FPS by default. I have tried adding it as a param to the launch file but that didn't work. I also tried adding it to PointGrey.cfg but that didn't work. I also tried adding it in the command line using "frame_rate:=30" but that didn't work either.

## Camera.launch ##
   
  <arg name="camera_serial" default="0" />
  <arg name="calibrated" default="0" />

  <group ns="camera">
    <node pkg="nodelet" type="nodelet" name="camera_nodelet_manager" args="manager" />

    <node pkg="nodelet" type="nodelet" name="camera_nodelet"
          args="load pointgrey_camera_driver/PointGreyCameraNodelet camera_nodelet_manager" >
      <param name="frame_id" value="camera" />
      <param name="serial" value="$(arg camera_serial)" />

      
      <param name="frame_rate" value="30" />
      
      
      <param name="camera_info_url" if="$(arg calibrated)"
             value="file://$(env HOME)/.ros/camera_info/$(arg camera_serial).yaml" />
    </node>

    <node pkg="nodelet" type="nodelet" name="image_proc_debayer"
          args="load image_proc/debayer camera_nodelet_manager">
    </node>
  </group>
</launch>

## PointGrey.cfg ##

    gen.add("frame_rate",         double_t, SensorLevels.RECONFIGURE_RUNNING,    "Camera speed (frames per second).",				                             30, 0, 100)

fabrizioschiano, 2017-02-07:
Hi, 

I know it is not the best way to reply to your question but for now I don't have time to write a full answer. I did write a kind of guide on how to set the resolution for a Flea Camera and this would help you also to set a frame rate at which the camera would start. It is handwritten , sorry for that.

Basically you need to _register_ your configuration in your camera. Look at the point 11 of my _guide_. 
(I suggest you to go through the whole document, it doesn't take much, you could learn something)

Let me know if you need further help and I will try to write that guide in better form.

[This is the link to the _guide_ ](https://www.dropbox.com/s/1mfzxmis4aslj62/ChangePointgreyFleaCameraResolution.pdf?dl=0)

davrsky, 2017-07-03:
Hi @fabrizioschiano the link is broken, do you have a newer version posted somewhere? Thanks

JaouadROS, 2017-10-16:
@fabrizioschiano Can we have the guide again please? or @bobjohnson2040 if you have it?

fabrizioschiano, 2017-10-25:
I am sorry for the delay, here is the guide:
[Change Pointgrey FLEA camera resolution.pdf](https://github.com/ros-drivers/pointgrey_camera_driver/files/1413889/Change.Pointgrey.FLEA.camera.resolution.pdf)