ros-drivers/pointgrey_camera_driver #96: cannot `list_cameras` because udev is not supported in docker

waspinator, 2017-02-13:
I'm trying to use docker with ROS, and I'm having trouble listing cameras because it appears to require udev. Is there a way I can use this package with Docker?

Thanks

adamreis, 2017-07-25:
@waspinator I'm running into the same problem. Did you have any luck getting this working?

mikepurvis, 2017-08-02:
Docker is isolated from hardware by design. Does it work if you mount the udev path inside the container? (eg, `-v /run/udev:/run/udev:ro`)

adamreis, 2017-08-04:
@mikepurvis thanks for following up on this. I'm pursuing another path, because I need to use Spinnaker with my Blackfly S, but I'll definitely give that a try and report back if I end up using this project.

waspinator, 2017-08-05:
@adamreis how do you plan on using Spinnaker with ROS?

vik748, 2017-10-12:
Anyone have any experience making Spinnaker work with ROS.  We have been trying for a while but seem to keep getting Segmentation Faults originating in the Spinnaker library when trying to instantiate Image Transport.  Our investigation shows that this is due to some conflict in Boost that both Spinnaker and Image Transport use.  GDB shows error coming from the Spinnaker Library.  If anyone has experience something similar or found a work around, please do share.
BTW we are trying to use the Blackfly S cameras.

mikepurvis, 2017-10-12:
@vik748 The Spinnaker so bundles in boost and a bunch of stuff, so I could totally imagine getting a bunch of weird symbol conflicts. A new spinnaker driver might need to run that in a separate process and pass images across shared memory to ROS. I'm not sure.

In any case, please don't resurrect this unrelated issue. If you'd like to discuss further, open a new one.

closed 2017-10-12 by mikepurvis