ros-drivers/pointgrey_camera_driver #100: Dynamic Reconfigure Parameter Limits lower than Hardware Limits

eitan-babcock, 2017-02-20:
I am trying to set the frame rate on my Point Grey Chameleon 3. The dynamic reconfigure has a max value of 100. I see that this is set in the PointGrey.cfg file. However, the camera itself is capable of higher frame rates.

I have installed the camera driver from apt:
`sudo apt-get install ros-kinetic-pointgrey-camera-driver`

Is there a way for me to modify the PointGrey.cfg file somewhere? I assume with my install method that it is all packaged up in a binary file and not changeable. Is there a way to override the default PointGrey.cfg file with a custom one? Can I point the Nodelet at a new .cfg file? Maybe in the launch file?

I have a custom launch file already where I set various parameters for the Nodelet. I generated this launch file by copying the camera.launch file from this github page and adding parameters. Is there a line I could add in here to tell the Nodelet to use a custom .cfg file?

I have tried installing from source so that I can modify the PointGrey.cfg file, but I have not had any luck getting that to work either. Are there instructions somewhere that could help me install this from source?

mikepurvis, 2017-02-20:
Building this package from source should be the same as building any other ROS package. What issues are you having?

In any case, please feel free to send a PR that expands the limits. This package is due for a new release soon anyways.

eitan-babcock, 2017-02-20:
I actually did get it installed from source now. I was doing it wrong. I was able to change that .cfg file and get higher frame rates. Thanks.

mikepurvis, 2017-02-21:
Please send a PR with the limits that work for your camera, so that the debs can be made to work for you and future users for your hardware.