ros-drivers/pointgrey_camera_driver #101: Flea3 outputs green image

jeff-o, 2017-02-22:
On two recently purchased USB3 Flea3 cameras, the image produced when using the default camera.launch file is always green-tinted. No matter what the start-up conditions (low light, high light, etc) the image starts and remains green.

We discovered that adding a parameter for auto_white_balance = false to the launch file seems to correct this behaviour. This never used to be necessary, so something must have changed in the hardware or driver. I don't mind adding this param into the launch file, but it's preferable if the issue were corrected at the root.

mikepurvis, 2017-02-22:
A PR related to this was recently merged (#95). Can you please try with the driver built from source and see if the issue remains?

stevendaniluk, 2017-03-16:
Since you haven't received any response, I'll mention that I was also experiencing the same green screen problem with a USB3 Blackfly camera , but building from source with the white balance changes updates PR #95 resolved it.

mikepurvis, 2017-03-16:
Excellent. Thanks, @stevendaniluk.

closed 2017-03-16 by mikepurvis