ros-drivers/pointgrey_camera_driver #105: Kinetic package missing from repo

crbarker, 2017-03-23:
The ROS wiki says to install as:
`sudo apt-get install ros-kinetic-pointgrey-camera-driver`

That worked a few days ago, but I think something must've broken recently, as the package is no longer found. Still seems to be there on indigo.

mikepurvis, 2017-03-24:
Yes, the 0.13.0 release failed to build and I haven't had a chance to roll it back. Feel free to submit such a PR to rosdistro (or troubleshoot the issue) if you're willing.

BenBlumer, 2017-03-29:
@mikepurvis How are you testing your builds? Is there a docker image or something?

For me, it seems to be working OK. I'm using the attached docker file. Building it as `docker build . -t pg_test`, running it with `docker-run -it pg_test` and then using `catkin_make`.

jonbinney, 2017-04-12:
Here is the log from a recent failed build on the buildfarm:

Looks like it is trying to `find_package` `roslaunch` and failing:
```
23:18:43 CMake Error at CMakeLists.txt:88 (find_package):
23:18:43   By not providing "Findroslaunch.cmake" in CMAKE_MODULE_PATH this project
23:18:43   has asked CMake to find a package configuration file provided by
23:18:43   "roslaunch", but CMake did not find one.
23:18:43 
23:18:43   Could not find a package configuration file provided by "roslaunch" with
23:18:43   any of the following names:
23:18:43 
23:18:43     roslaunchConfig.cmake
23:18:43     roslaunch-config.cmake
23:18:43 
23:18:43   Add the installation prefix of "roslaunch" to CMAKE_PREFIX_PATH or set
23:18:43   "roslaunch_DIR" to a directory containing one of the above files.  If
23:18:43   "roslaunch" provides a separate development package or SDK, be sure it has
23:18:43   been installed.
23:18:43 
23:18:43 
23:18:43 -- Configuring incomplete, errors occurred!
23:18:43 See also "/tmp/binarydeb/ros-kinetic-pointgrey-camera-driver-0.13.0/obj-x86_64-linux-gnu/CMakeFiles/CMakeOutput.log".
23:18:43 See also "/tmp/binarydeb/ros-kinetic-pointgrey-camera-driver-0.13.0/obj-x86_64-linux-gnu/CMakeFiles/CMakeError.log".
23:18:43 	cd /tmp/binarydeb/ros-kinetic-pointgrey-camera-driver-0.13.0
23:18:43 	cd obj-x86_64-linux-gnu
23:18:43 	"tail -v -n +0 CMakeCache.txt"
```

This is caused by line 88 in pointgrey_camera_driver's cmakelists, which is:
```
if (CATKIN_ENABLE_TESTING)
88	  find_package(roslaunch REQUIRED)
89	  roslaunch_add_file_check(launch/bumblebee.launch)
```

I've never used this find_package roslaunch and then adding roslaunch check files, so I'm not sure if that is supposed to work.

jonbinney, 2017-04-13:
(edited above comment to add link to the actual buildfarm log)

jonbinney, 2017-04-13:
@tonybaltovski - looks like those cmake lines were added in your commit: https://github.com/ros-drivers/pointgrey_camera_driver/commit/648ab6224958ba705377936c5de02e2136c1a656

Any idea why they aren't working on the buildfarm?

mikepurvis, 2017-04-13:
I added the missing dependency in 9d4c850ca374fa3605336b53383a436d95bd9741, but the devel job is still failing on account of some other changes that I cherry-picked in:

http://build.ros.org/job/Kdev__pointgrey_camera_driver__ubuntu_xenial_amd64/10/console

mikepurvis, 2017-04-13:
The problem change is 07ea31940b4a4117d67e82d7928a3d5e3ad74e09, but the error is bizarre:

```
-- catkin 0.7.6
-- libflycapture found in system library path
-- libflycapture library: /tmp/catkin_workspace/devel_isolated/pointgrey_camera_driver/lib/libflycapture.so.2
-- libflycapture copied from : /tmp/catkin_workspace/devel_isolated/pointgrey_camera_driver/lib/libflycapture.so.2 into /tmp/catkin_workspace/build_isolated/pointgrey_camera_driver//tmp/catkin_workspace/devel_isolated/pointgrey_camera_driver/lib/libflycapture.so.2
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/catkin_workspace/build_isolated/pointgrey_camera_driver
==> make -j1 in '/tmp/catkin_workspace/build_isolated/pointgrey_camera_driver'
[  7%] Generating dynamic reconfigure files from cfg/PointGrey.cfg: /tmp/catkin_workspace/devel_isolated/pointgrey_camera_driver/include/pointgrey_camera_driver/PointGreyConfig.h /tmp/catkin_workspace/devel_isolated/pointgrey_camera_driver/lib/python2.7/dist-packages/pointgrey_camera_driver/cfg/PointGreyConfig.py
Generating reconfiguration files for PointGrey in pointgrey_camera_driver
Wrote header file in /tmp/catkin_workspace/devel_isolated/pointgrey_camera_driver/include/pointgrey_camera_driver/PointGreyConfig.h
[  7%] Built target pointgrey_camera_driver_gencfg
Scanning dependencies of target PointGreyCamera
[ 15%] Building CXX object CMakeFiles/PointGreyCamera.dir/src/PointGreyCamera.cpp.o
In file included from /tmp/catkin_workspace/src/pointgrey_camera_driver/pointgrey_camera_driver/src/PointGreyCamera.cpp:32:0:
/tmp/catkin_workspace/src/pointgrey_camera_driver/pointgrey_camera_driver/include/pointgrey_camera_driver/PointGreyCamera.h:46:36: fatal error: flycapture/FlyCapture2.h: No such file or directory
compilation terminated.
```

How is it finding libflycapture in the devel space during the configure step, prior to building anything?

jonbinney, 2017-04-13:
Yeah i've been having the same problem (it doesn't find flycapture sometimes) locally when using master branch. Although it has always built for me the first time; on successive builds it fails.

jonbinney, 2017-04-13:
Its got to be https://github.com/ros-drivers/pointgrey_camera_driver/commit/07ea31940b4a4117d67e82d7928a3d5e3ad74e09 causing the problem, right? Is it worth reverting that for now to get a working release and then re-adding and debugging it later?

mikepurvis, 2017-04-13:
Worst case I can just revert that change, but it's helpful for our internal build process at Clearpath, and I'd like to get us back on upstream with this repo for the purposes of testing a common codebase.

jonbinney, 2017-04-13:
Ah I see. +1 for having the maintainer being able to use the upstream :-)

jonbinney, 2017-04-13:
That cmake logic only matters at build time, right? So what about reverting that commit, releasing debs, then re-applying that change, and we can debug before the next release? That way the master branch should work for you guys, and the deb should work.

mikepurvis, 2017-04-13:
Oh blah. I see the problem. I'll have a fix up shortly.

mikepurvis, 2017-04-13:
I think the issue should be resolved with 432ca464cc2d170072e4f7ed57578414eb06a481. I've confirmed it works on local build and re-build (which was the issue before).

@dirk-thomas Can you trigger http://build.ros.org/job/Kdev__pointgrey_camera_driver__ubuntu_xenial_amd64/ for me?

jonbinney, 2017-04-13:
Thanks @mikepurvis !

dirk-thomas, 2017-04-13:
http://build.ros.org/job/Kdev__pointgrey_camera_driver__ubuntu_xenial_amd64/13/

mikepurvis, 2017-04-13:
This failure makes even less sense to me than the previous one— seems like it's dying now in a buildfarm infrastructure step, before it even gets to any building:

```
Get base image name from Dockerfile 'Dockerfile': ubuntu:xenial
Check docker base image for updates: docker pull ubuntu:xenial
xenial: Pulling from library/ubuntu
Digest: sha256:10f6cc4e82187a4c57e2268453b7d8b5303740b4cf87cfbb425c28751809b79d
Status: Image is up to date for ubuntu:xenial
+ docker build --force-rm -t devel_task_generation.kinetic_pointgrey_camera_driver .
Sending build context to Docker daemon 15.36 kB
Sending build context to Docker daemon 15.36 kB

Step 1 : FROM ubuntu:xenial
 ---> 0156b919f2e5
Step 2 : VOLUME /var/cache/apt/archives
 ---> Using cache
 ---> 1c79db7e9832
Step 3 : ENV DEBIAN_FRONTEND noninteractive
 ---> Using cache
 ---> 3b270241c1ad
Step 4 : RUN locale-gen en_US.UTF-8
 ---> Running in f3ad0422a881
[91m/bin/sh: 1: locale-gen: not found
[0mRemoving intermediate container f3ad0422a881
The command '/bin/sh -c locale-gen en_US.UTF-8' returned a non-zero code: 127
Build step 'Execute shell' marked build as failure
```

@dirk-thomas Do you know what's going on here?

dirk-thomas, 2017-04-13:
Please see ros-infrastructure/ros_buildfarm#415

mikepurvis, 2017-04-13:
Cool, thanks!

mikepurvis, 2017-04-19:
Release up: https://github.com/ros/rosdistro/pull/14571

closed 2017-04-19 by mikepurvis