ros-drivers/pointgrey_camera_driver #108: Pointgrey_camera_driver crashes when topics are echoed

arjun-narang, 2017-04-12:
Hello,

I am using Chameleon3 CM3-U3-50S5C with the pointgrey_camera_driver. The camera.launch file runs successfully but when I do a rostopic echo on the camera image the process dies and throws the following error:

[camera/camera_nodelet_manager-2] process has died [pid 3777, exit code -11, cmd /opt/ros/indigo/lib/nodelet/nodelet manager __name:=camera_nodelet_manager __log:=/home/rj/.ros/log/0b0cf670-1f15-11e7-8cc6-e4a7a033c980/camera-camera_nodelet_manager-2.log].
log file: /home/rj/.ros/log/0b0cf670-1f15-11e7-8cc6-e4a7a033c980/camera-camera_nodelet_manager-2*.log

Can someone please help me with this?

Few more things: 
Flycapture works fine.
I am able to detect the camera
I have set the necessary permissions.

jonbinney, 2017-04-12:
"exit code -11" means a segmentation fault. There are a couple possible causes for that, but with dynamically loaded libraries (like nodelets) involved, I tend to suspect that different things (exectuables/objects) have probably been compiled with different versions of something that has ABI breaking changes.

Did you install using apt-get? Can you do a `sudo apt-get update` and `sudo apt-get dist-upgrade` to make sure you have the newest version of everything? Then completely rebuild anything you are using from source, and try running things again.

arjun-narang, 2017-04-14:
Thanks. Fixed it myself.

closed 2017-04-14 by arjun-narang