ros-drivers/pointgrey_camera_driver #113: Kinetic package missing from repo

gkguru, 2017-04-24:
I tried ``apt-get install ros-kinetic-pointgrey-camera-driver`` to install the package but got the following error:
```
E: Unable to locate package ros-kinetic-pointgrey-camera-driver
```

gkguru, 2017-04-24:
``apt-cache search pointgrey`` yields only ``ros-kinetic-pointgrey-camera-description``
```
ros-kinetic-pointgrey-camera-description - A series of URDF driver decriptions for Point Grey cameras
```

mikepurvis, 2017-07-18:
Things are looking happier now: http://repositories.ros.org/status_page/ros_kinetic_default.html?q=pointgrey

closed 2017-07-18 by mikepurvis

JustBenj, 2019-06-08:
Missing again as of 8pm June 7, 2019. Same results as the original issue. Is it related to https://discourse.ros.org/t/new-gpg-keys-deployed-for-packages-ros-org/9454?

rubenanapu, 2019-06-10:
As of today, 10th of June, 2019, the error is happening again.

djhanove, 2019-06-16:
I'm having these same issues when i use the command apt-get install ros-kinetic-pointgrey-camera-driver.
I'm using 16.04, x86

thefinaljob, 2019-06-28:
Missing again as of June 27th, 2019, can someone fix this asap?

thefinaljob, 2019-06-28:
@mikepurvis

alvikag, 2019-09-13:
I am facing the same issue. I am using ros kinetic, ubuntu 16.04

mtfurlan, 2019-11-11:
This issue came up higher in google, so for for the next person to come along:
Look at #187 for the reason and #188 for a janky snapshot workaround.