ros-drivers/pointgrey_camera_driver #114: Unable to store image data as rosbag from stereo cameras simultaneously.

ashnarayan13, 2017-05-05:
I want to store the image data from the cameras in real time. When I run the rosbag record, one of the camera nodes closes. 
Any suggestions?

mikepurvis, 2017-05-05:
The topics are transcoded on demand, so if you try to subscribe to all of them, you'll peg your CPU. Did you record a specific topic, or just `rosbag record -a`?

ashnarayan13, 2017-05-05:
I am able to record a single topic at a time. I want to record the image_raw from both left and right cams simultaneously. When I try that, it crashes the right cam always. I am using flea3 cams. 
rosbag record -a stops one of the cams from recording (right cam) and only left cam data is available

naveenT1010, 2017-07-26:
Hi @mikepurvis, I am also trying to record two camera topics, but most of the times, one of the camera nodelet dies. Even if i try to view the images in rqt (both the cameras simultaneously) it again kills one of the camera nodelet.

ashnarayan13, 2017-11-14:
This can be fixed by changing the buffer for the USB on Jetson tx1 (device I was using). 
Also, reducing the framerates / image resolution can help with this.

closed 2017-11-14 by ashnarayan13

BoomFan, 2018-05-10:
Would you mind sharing the way you change USB buffer, please? I'm running into the same issue. And not only on Jetson tx1, but also on my personal laptop.