ros-drivers/pointgrey_camera_driver #117: Node crashes after about 30 seconds

k-kleber, 2017-05-08:
Hello,
My camera node keeps crashing after about 30 seconds. Here is the output:
`SUMMARY
========

PARAMETERS
 * /camera/camera_nodelet/frame_id: camera
 * /camera/camera_nodelet/serial: 0
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

NODES
  /camera/
    camera_nodelet (nodelet/nodelet)
    camera_nodelet_manager (nodelet/nodelet)
    image_proc_debayer (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [5030]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 780e3fc0-3407-11e7-bd21-94de806f2142
process[rosout-1]: started with pid [5043]
started core service [/rosout]
process[camera/camera_nodelet_manager-2]: started with pid [5050]
process[camera/camera_nodelet-3]: started with pid [5061]
process[camera/image_proc_debayer-4]: started with pid [5062]
[ERROR] [1494259292.179409307]: PointGreyCamera::getCameraTemperature Could not get property. | FlyCapture2::ErrorType 28 Feature register read failed.
[ERROR] [1494259292.281687957]: Failed to disconnect with error: PointGreyCamera::disconnect Failed to disconnect camera | FlyCapture2::ErrorType 35 Error stopping stream.`

After this I can close it using ctrl-c and restart it. Then the same happens again after about 30 seconds.

I am using Ubuntu 16.04 and have the same issue when using the apt repository and the github's master branch.

Please tell me what additional information you might need.
Thank you for the help.

apekshaK, 2017-06-21:
I am getting the same issue... were you able to find a solution?

k-kleber, 2017-06-29:
It is a camera issue I think. When using flycapture it also crashes without any errors. According to Pointgrey it is a connection issue, but we could not completely solve it. Using different USB cables and making sure, that these are firmly attached can help.

closed 2017-06-29 by k-kleber