ros-drivers/pointgrey_camera_driver #118: possible Spinnaker / GenICam3 compatiblity?

waspinator, 2017-05-08:
Point Grey is releasing their new models using the Spinnaker with GenICam3 support rather than flycapture2. Any thoughts on supporting the new sdk?

mikepurvis, 2017-05-29:
Take a stab at it. If the implementation looks okay, we'd consider merging. If it's a mess, then a separate package is probably the most logical.

mrouea, 2018-03-07:
Hi Mike, 

I just received a blackfly S camera, which is working with the spinnaker SDK. I wonder that the pointgrey_camera_driver is not working with this SDK since flycap SDK is requested. 

My question: did you already do the merging of the ros package in order to take into account the new SDK? 
Pointgrey starts releasing the new camera model with spinnaker SDK, and many new cameras will not be considered in your ros package. 

Thank you in advance for your support, 

Ali

shahvi, 2018-07-13:
@mrouea our lab has just released our drivers for the spinnaker based cameras at https://github.com/neufieldrobotics/spinnaker-ros , please do give it a shot.