ros-drivers/pointgrey_camera_driver #121: PointGrey Blackfly Gige camera, no camera detected

Ianlovescode, 2017-05-25:
Hi all,

I am using PointGrey Blackfly Gige camera, and i run rosrun pointgrey_camera_driver list_cameras, i was given that there is no camera detected  on computer. I couldn`t run camera driver as well.
After that, i used FlyCap2 software and used auto force ip to find camera. Then run list_cameras again, everything was working. 

Is it the correct way to find camera?

Thanks.

yxh0606, 2017-10-04:
Hi lanlovecode,

Currently I use the Ubuntu 16.04 64 bit with ros kinect distributions.
By the way, we are using the FLIR(PointGrey) Blackfly Gige camera(BFS-PGE series) using Ethernet port(not usb port).     
As for your knowledge, FLIR has inked a definitive asset purchase agreement to buy Point Grey Research, Inc.in 2016.

Same issues with you. Camera can't be detected by  run rosrun pointgrey_camera_driver list_cameras.

I can ping the camera IP address, that means it works well.  But when I run the flycap software in terminal, it can't detect the Gige camera. At the same time, it still can't find the Gige Camera by rosrun pointgrey_camera_driver list_cameras. 

Is this the SDK not compatible with Gige camera using Ethernet port? 
Is this some command to run the camera that we can use flycap to detect it? 

Thank you very much and wait for your response.

DrCurry, 2018-07-24:
BFS cameras only supports Spinnaker. pointgrey_camera_driver uses FlyCapture.

DrCurry, 2018-07-24:
@Ianlovescode and @yxh0606 are you using Spinnaker or FlyCapture SDKs? Also what model for the camera? The pointgrey_camera_driver uses FlyCapture.

IceSentry, 2018-08-30:
Did you find a workaround? I have a similar issue. I can ping the camera, but the driver doesn't see it

yxh0606, 2018-08-31:
Dear IceSentry

Still have the issue that it can't detect the camera. 
Actually we modify the drive code by ourself. 
You might wait the FLIR camera that it officially launch the ROS package for the camera.  

Many thanks.

addext, 2019-07-29:
same issue - did anyone figure it out?

IceSentry, 2019-07-29:
We just threw the camera away and bought a cheap usb webcam

addext, 2019-07-30:
that's a pity. We paid good money for them and all the accessories.

thanujarc, 2019-09-23:
Hi, 

I have a similar issue. I have two BLFY-PGE-31S4C-C GiGE PoE cameras which are connected to a Linux system (A Jetson TX2) via a PoE GiGE Netgear switch. However, I cannot get the TX2 to recognize the cameras. flycap software shows no detected devices, and auto force ip doesn't return the details of any of the connected cameras.    

Can someone please let me know which IP address I should be using for the host network interface? How can find out the IP addresses of the cameras?

Thank you.

yxh0606, 2019-09-23:
Dear thanujarc,

I encountered this issue before in Ubuntu 16.04 Operating System. 
You could download the Windows Spinnaker software from official website. 

https://www.flir.asia/support-center/iis/machine-vision/downloads/spinnaker-sdk-flycapture-and-firmware-download/

Then install it in window OS, then could detect and record the ip address for camera. 

If have the ip address, configure the necessary ip settings for you Linux system. And it could detect the IP camera. 

Please feel free to contact me for further discussion. Many thanks.

Best Regards,
Xuehuan
[image]

chris-grebe, 2019-11-29:
> Dear thanujarc,
>
> I encountered this issue before in Ubuntu 16.04 Operating System.
> You could download the Windows Spinnaker software from official website. [...]

Have you had any experience using Linux with Point Grey Flea 3? Running the following command:
`rosrun pointgrey_camera_driver list_cameras`
from `http://wiki.ros.org/pointgrey_camera_driver` with the camera attached, but no cameras are detectable. The camera light is yellow or possibly yellow-green which means "initializing FPGA" and "camera power down" respectively according to the datasheet.

yxh0606, 2019-11-29:
Dear Thanujarc,

You could install following code in the GitHub.

https://github.com/KumarRobotics/flea3

Many thanks.

Best Regards,
Xuehuan

Soodep, 2020-11-05:
i am still having the same issue. did anyone solve this problem? asking for help

Soodep, 2020-11-05:
> Dear thanujarc,
>
> I encountered this issue before in Ubuntu 16.04 Operating System.
> You could download the Windows Spinnaker software from official website. [...]

I followed your instruction and got the IP address from my windows spinview, now how do i configure the ip address in Ubuntu, would you please provide a clear instruction, i would apppreciate it