ros-drivers/pointgrey_camera_driver #123: fatal error: pointgrey_camera_driver/PointGreyCamera.h: No such file or directory

benjaminabruzzo, 2017-06-21:
I just asked a question over at answers.ros, I figured I would repeat it here just in case someone else has already solved the issue:  [http://answers.ros.org/question/264329/fatal-error-pointgrey_camera_driverpointgreycamerah-no-such-file-or-directory/](question)

I am trying to set registers on a camera from another node in my package after the cameras have already been launched.  I've done this with 1394 cameras and am trying to follow the same path for two GigE cameras (2x blackfly 09s2c).  I think it is a cmake issue, which is why I asked it as ROS primarily instead of here.

Thanks in advance.

davrsky, 2017-07-03:
Having similar issues with multiple gige cameras. If I start the nodes one at a time with a long delay inbetween I can usually get it to work for a while, but they still produce intermittent crashes. Packet delay and frame sizes have been set that I am nowhere near hitting the 1 gigabit limit (it's around 450mbit/s right now when they are all working). And even if it were a bandwidth issue, a dropped frame should not cause the camera driver to crash. I've also set the sysctl as recommended by point grey & in other issues here, and I have made sure that the camera nodes are in different namespaces with different node names as well, as that was mentioned in another issue.

Have you had any luck solving this?

benjaminabruzzo, 2017-07-04:
I have not.  I am currently using two Chameleon cameras as a stop-gap.

paulashbourne, 2017-09-14:
@davrsky @benjaminabruzzo Just wanted to follow up on this thread and see if either have you have found a resolution. Been experiencing this bug for a couple months on my GigEs.

davrsky, 2017-09-14:
I ended up having to dedicate one ethernet port per camera, was the only
way I could make the point greys operate reliably

benjaminabruzzo, 2017-09-14:
Negative.  I am still using the chameleons.  

I tried using a router between the PC and the cameras as well as plugging them in directly to the two ethernet ports.  

@davrsky What kind of configuring did you need to do to set up the 2x ports?