ros-drivers/pointgrey_camera_driver #128: Software trigger for GIGE camera

ashnarayan13, 2017-07-09:
Is there a particular setting to set the trigger based on software? 

I have gone through the cfg file provided with the ROS package. But I am not sure which setting is for the software delay or if it can be implemented. 

Pointgrey says that asynchronous trigger can be set using software "All cameras that implement a software asynchronous trigger can be triggered via the SOFTWARE_TRIGGER register 0x62C". 

Is this possible with the ros package? 

Thanks!

mjavault, 2018-07-18:
@tonybaltovski I had the same need and updated the code to add software trigger support. What's the proper way of sending a PR for this?

tonybaltovski, 2018-07-19:
@mjavault the way you did it is proper.  The maintainer will review it when they get a chance.