ros-drivers/pointgrey_camera_driver #134: */camera_info not publishing

benjaminabruzzo, 2017-07-19:
I am having an issue where the camera info is not getting published when running the driver.  The images are getting published, just not the camera info.  This is an issue with using other packages that require the camera info.

Does anyone have any idea why this might be the case?
(yaml and launch files changed to txt to upload)

[17094423yaml.txt](https://github.com/ros-drivers/pointgrey_camera_driver/files/1160763/17094423yaml.txt)

[create_pgrlaunch.txt](https://github.com/ros-drivers/pointgrey_camera_driver/files/1160764/create_pgrlaunch.txt)

[16369047yaml.txt](https://github.com/ros-drivers/pointgrey_camera_driver/files/1160759/16369047yaml.txt)

benjaminabruzzo, 2017-07-20:
I'm not sure what the collision is, but I have isolated the issue to installing the viso2 visual odometry package: http://wiki.ros.org/viso2_ros

mikepurvis, 2017-07-20:
Is it enough to simply _install_ that package without running anything? If so, that's super weird. What's the output of:

```
dpkg -l ros-indigo-viso2-ros
dpkg -l ros-indigo-viso2
dpkg -l ros-indigo-libviso2
```

Just wondering if one these is smuggling a bad udev rule or something else into your system.

Does the problem only manifest when you're using the package, or is simply installing it enough?

Under other circumstances, I would have assumed outdated firmware on your cameras, but the connection to this other package is super screwy.

benjaminabruzzo, 2017-07-21:
I'll check those in a moment, I'm also working on a minimal clean install and a minimal launch file for the same issue opened on the viso2 git issue page.

benjaminabruzzo, 2017-07-22:
I think the issue that I'm having is how the driver handles USB cameras.  Using a pair of USB3 chameleons (cm3-us-13s2c), a pair of USB2 fireflies (mv fmvu-03mtc), and a pair of gigE blackflies (bfly-pge-13h2c), I tested this driver for all three pairs, and on the fireflies, I tested the 1394 driver.  The four launch files are linked below.

Using this driver, the blackflies are able to publish both images and their camera info (though when fired by an external trigger simultaneously, the image consistency errors cause dropped frames). 

Using this driver, the chameleons publish the images, but not the camera_info.

Lastly, using this driver, the fireflie publish the images, but not the camera_info.  However, using the 1394 driver, both images and camera_info are getting published.

To install the 1394 driver, I used these commands:
```
sudo apt-get install ros-indigo-camera1394
rosdep install camera1394
```

[files](https://drive.google.com/open?id=0By_OMkAs76xHUWcyb1A1Z21UeGM)

benjaminabruzzo, 2017-07-24:
By using the apt-get installation in ros-indigo, I may have just duplicated the symptoms from [this gist](https://gist.github.com/rossbar/ebb282c3b73c41c1404123de6cea4771)

benjaminabruzzo, 2017-07-25:
For others: I modified the file in the above gist to use a ros parameter for the yaml filename (instead of a command line argument) and then made a launch file for it:
[node_py](https://github.com/benjaminabruzzo/pgr_mwe/blob/master/launch/yaml_info_pub.py)
[launch](https://github.com/benjaminabruzzo/pgr_mwe/blob/master/launch/info.launch)

closed 2017-07-25 by benjaminabruzzo