ros-drivers/pointgrey_camera_driver #135: Problem using sudo apt-get install ros-kinetic-pointgrey-camera-driver

lingyee, 2017-07-20:
I tried sudo apt-get install ros-kinetic-pointgrey-camera-driver to install the package. Installation is successful but when I tried to launch Point Grey's FlyCapture2 Camera Selection through command line, it launches version 2.9.3.43 instead of 2.11.3.121, then the software crashes when it tries to capture. It was version 2.11.3.121 before install from apt. I'm using Ubuntu 16.04 LTS.

mikepurvis, 2017-07-20:
I've just merged ef74de62985baa4ad83e7add38968b10701c8538, which bumps the included version of flycapture. A release will be forthcoming shortly.

varunagrawal, 2017-09-27:
@mikepurvis any updates on the release?
@a-price

mikepurvis, 2017-09-28:
Thanks for the reminder!

Kinetic: https://github.com/ros/rosdistro/pull/15981

Lunar: https://github.com/ros/rosdistro/pull/15982