ros-drivers/pointgrey_camera_driver #136: Hardware Synchronization with multiple cameras

stormtiti, 2017-07-21:
1) Ubuntu 16.04 LTS AMD64 with i7 4 core
2) ROS Kinetic Version
3) pointgrey_camera_driver git clone from source code
4) cameras connected with GPIO cables
5) use FlyCap2 GUI tools to setup "Hardware Synchronization" and it works well
6) setup ROS launch file, if each camera has no "Synchronization" config and I can use rqt can fetch image data on the same time.
-----------
Problem:
7) Now setup master and slave cameras for Hardware Synchronization, it doesn't work (can only fetch master camera data) Even I try to use rqt_reconfiguration, it doesn't work ~~
-----------
Here is may question:
1) can this open source driver support hardware synchronization and How ? 
THX for code authors ~~

stormtiti, 2017-07-21:
BTW I use PointGrey GS3 USB 3.0

stormtiti, 2017-07-24:
[image]

Here the warning information ~~

srsidd, 2019-01-24:
@stormtiti I know it's been a while, but do you have any information you can provide about this?