ros-drivers/pointgrey_camera_driver #143: could not discover packet_size

zfxw0206, 2017-09-15:
Some times, when I start my  Blackfly BFLY-PGE-23S2C, I received "PointGreyCamera::connect could not discover GigE packet_size | FlyCapture2::ErrorType 1Failed to discover GigE packet size", even though I tried to restart the driver, the problem is still there. I checked the driver, it comes from "DiscoverGigEPacketSize" function from flycapture. 

Is there anybody cam tell me how does "DiscoverGigEPacketSize" works and how to recovery this error.

Thank you!

sheelabhadra, 2017-10-27:
Hi @zfxw0206 , I am facing a similar issue with a BlackFly BFLY-PGE-20E4C camera. Were you able to solve the issue?

Thanks!

sheelabhadra, 2017-10-27:
I didn't get the error anymore after I closed the FlyCapture application and ran the launch file in the terminal.

zfxw0206, 2017-11-03:
Some times I still have the issue after I power cycle the whole system(including the PC, switch and camera)