ros-drivers/pointgrey_camera_driver #144: Support for Ubuntu Yakkety, Zesty and Debian stretch

mikaelarguedas, 2017-10-02:
Howdy @mikepurvis and pointgrey_camera_driver maintainers.

It seems that this package fails to build on Lunar on all platforms except ubuntu Xenial.
At a glance it looks like the [download_flycap script](https://github.com/ros-drivers/pointgrey_camera_driver/blob/master/pointgrey_camera_driver/cmake/download_flycap) needs to be extended for more platforms if possible.

mikepurvis, 2017-10-03:
I think the upstream SDK from the vendor only explicitly supports the Ubuntu LTS versions, but it's also an extremely large `so` which seems to have its dependencies bundled, so it probably would work just fine.

If we're to modify the download script with a mapping that allows it to treat all these other platforms as if they were Xenial, what should the list of them be? Based on REP 3 required+recommended and the fact that http://jenkins.ros.org/ doesn't build for Fedora, that would be yakkety, zesty, stretch, and jessie.

mikaelarguedas, 2017-10-03:
Indeed we don't build binaries for fedora, so the list of platforms is the one you listed. Thanks for the quick fix!

PR #146 (2017-10-03): Assume current library version for all OSes other than trusty.
This should resolve #144.

FYI @mikaelarguedas @tfoote
closed 2017-10-03 by mikepurvis