ros-drivers/pointgrey_camera_driver #145: pointgrey_camera_driver is failing to build on Debian Jessie

tfoote, 2017-10-03:
It looks like regression after: https://github.com/ros/rosdistro/pull/15981

http://build.ros.org/view/Kbin_dj_dJ64/job/Kbin_dj_dJ64__pointgrey_camera_driver__debian_jessie_amd64__binary/

The failure is:
```
01:19:58 In file included from /tmp/binarydeb/ros-kinetic-pointgrey-camera-driver-0.13.2/src/PointGreyCamera.cpp:32:0:
01:19:58 /tmp/binarydeb/ros-kinetic-pointgrey-camera-driver-0.13.2/include/pointgrey_camera_driver/PointGreyCamera.h:46:36: fatal error: flycapture/FlyCapture2.h: No such file or directory
01:19:58  #include "flycapture/FlyCapture2.h"
01:19:58                                     ^
```

The referenced commit (https://github.com/ros-drivers/pointgrey_camera_driver/commit/ef74de62985baa4ad83e7add38968b10701c8538) in #135  doesn't seem to call out Debian Jessie in the listings.

Before this last release Jessie was building cleanly. http://build.ros.org/view/Kbin_dj_dJ64/job/Kbin_dj_dJ64__pointgrey_camera_driver__debian_jessie_amd64__binary/

Though before the Debian Jessie on arm64 wasn't building: http://build.ros.org/view/Kbin_djv8_dJv8/job/Kbin_djv8_dJv8__pointgrey_camera_driver__debian_jessie_arm64__binary/

mikaelarguedas, 2017-10-03:
symptoms seem similar to #144. hopefully calling out all targeted platforms will fix both this and #144

tfoote, 2017-10-03:
Yes, sorry I missed that due to non-overlapping keywords. It definitely looks like the same issue.

mikepurvis, 2017-10-03:
Closing as dupe.

closed 2017-10-03 by mikepurvis