ros-drivers/pointgrey_camera_driver #147: Jessie builds of pointgrey_camera_driver fail to build

tfoote, 2017-10-07:
This is a follow up to https://github.com/ros-drivers/pointgrey_camera_driver/pull/146#pullrequestreview-66666240 

jessie amd64 has regressed: http://build.ros.org/view/Kbin_dj_dJ64/job/Kbin_dj_dJ64__pointgrey_camera_driver__debian_jessie_amd64__binary/64/console

and

jessie arm64 continues to fail with the same issue: http://build.ros.org/view/Kbin_djv8_dJv8/job/Kbin_djv8_dJv8__pointgrey_camera_driver__debian_jessie_arm64__binary/371/console

mikepurvis, 2017-10-19:
Should be resolved by ae67188a5b49b56b1d5d5b102301f7d8a3523a92. Will need a new release, though.

tannernray, 2017-11-01:
Has this new release occurred yet?

mikepurvis, 2017-11-01:
Yes, `0.13.4`. It's building successfully now:

http://build.ros.org/view/Kbin_dj_dJ64/job/Kbin_dj_dJ64__pointgrey_camera_driver__debian_jessie_amd64__binary/

closed 2017-11-01 by mikepurvis

tannernray, 2017-11-01:
Weird, I'm getting the same error after installing the newest from source

> No such file or directory
 #include "flycapture/FlyCapture2.h"

ChunJyeBehBeh, 2018-04-06:
Hello I face the problem also when i catkin_make it on Raspberry pi Jessie. May I know how to solve the problem? I struggle it for a long time already