ros-drivers/pointgrey_camera_driver #148: Unable to locate package ros-lunar-pointgrey-camera-driver

LifeBeyondExpectations, 2017-10-11:
I do not know why I cannot install pointgrey-driver

.. Unable to locate package ros-lunar-pointgrey-camera-driver .. 

`$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.2 LTS
Release:	16.04
Codename:	xenial
`

Of course Flycapture2 has been installed well.
I would like to connect FL2G-13S2C-C (pointgrey camera)  to my computer on ROS

mikaelarguedas, 2017-10-11:
@LifeBeyondExpectations `ros-lunar-pointgrey-camera-driver` has been [synced today](https://discourse.ros.org/t/new-packages-for-lunar-2017-10-11/2909) to the main repository. Packages usually stay ~2weeks in testing before being made available on the main repository.

Can you please update your sources and try to install it again ?
Thanks!

sbrahma0, 2019-07-29:
Hey, i tried to install the point grey package using sudo apt-get install ros-kinetic-pointgrey-camera-driver
but its showing- Unable to locate package ros-kinetic-pointgrey-camera-driver.
Can you please help me out with this.

anakin0620, 2019-12-03:
@sbrahma0  Once I figured out I will let you know.

anakin0620, 2019-12-03:
@sbrahma0  https://github.com/ros-drivers/pointgrey_camera_driver/issues/188#issuecomment-521527548