ros-drivers/pointgrey_camera_driver #151: FlyCapture2::ErrorType 33

LEEluoman, 2017-10-29:
Hello everyone.
My system version is ubuntu 16.04 and Ros version is kinetic 
when I run command " rosrun pointgrey_camera_driver list_cameras"
>  Number of cameras found: 1
[0]Serial: 15399215, Model: Firefly MV FMVU-03MTM, Vendor: Point Grey Research, Sensor: Micron MT9V022177ATM (1/3" 640x480 CMOS), Resolution: 752x480, Color: false, Firmware Version: 1.8.3.0

but when I run "roslaunch pointgrey_camera_driver camera.launch" and try to echo the camera message "/camera/image_raw" or show it in rviz, I get the error:
>  Failed to start with error: PointGreyCamera::start Failed to start capture | FlyCapture2::ErrorType 33 Error starting isochronous stream.

And I raise the amount of memory allocated, although my camera is USB2.0, then i run "cat /sys/module/usbcore/parameters/usbfs_memory_mb"
> 16

I can get camera image in the FlyCap2 software, but I need get camera image in ros.
I need help.
Thank you!

fnoble, 2017-10-31:
I had this same issue. It seems to be related to the dependencies pulled in by installing the `ros-kinetic-pointgrey-camera-driver` from the Ubuntu package.

I installed FlyCapture2 from Point Grey and this worked but when `ros-kinetic-pointgrey-camera-driver` was installed `flycap` stopped working and I got the same errors as above when using `roslaunch`. I uninstalled `ros-kinetic-pointgrey-camera-driver` including doing an autoremove of the orphaned packages and `flycap` started working again.

I installed `pointgrey_camera_driver` from source and it pulled in the FlyCapture2 libraries that were already on my system from the Point Grey installer and now works fine.

royaalto, 2017-11-27:
@fnoble could you give me more details about how you do that , i am new to this, thx.

BoomFan, 2018-05-11:
I have the same issue with the stereo camera launch file "stereo.launch". Since I pull the source code directly from Github, uninstalling `ros-kinetic-pointgrey-camera-driver` didn't help.
Did you guys find a solution yet?

a137755952, 2018-11-12:
Is there anyone has solved the ErrorType 33 problem yet? @LEEluoman @royaalto I also get the error when I try to get image from Pt. camera.

shi-yan, 2019-01-31:
same here, I don't even have ros.

shi-yan, 2019-01-31:
well, I did this:

sudo sh -c 'echo 1024 > /sys/module/usbcore/parameters/usbfs_memory_mb'

cat /sys/module/usbcore/parameters/usbfs_memory_mb

it worked now.