ros-drivers/pointgrey_camera_driver #155: Reconfigure Callback failed with error: PointGreyCamera::SetGigEConfig could not set GigE settings

MichaelWang1028, 2017-12-01:
I'm using ubuntu16.04 and a kinect version ROS, I install using the command: sudo apt-get install ros-kinect-pointgrey-camera-driver.

when I using the command: rosrun pointgrey_camera_driver camera.launch, this error occored:

[ERROR] [1512118059.821921194]: Reconfigure Callback failed with error: PointGreyCamera::SetGigEConfig could not set GigE settings (packet resend) | FlyCapture2::ErrorType 7 Packet resend is not supported.

xubo0321, 2018-06-22:
@MichaelWang1028  
did you solve the problem?

Sindorman, 2018-06-22:
@xubo0321 Yes, It is solved in #163. I have compiled their code locally, edited launch file for packet resend argument and it works well. They need to merge it.

xubo0321, 2018-06-22:
@Sindorman how to merge it ,what i can do,can you tell it in details? thx.

xubo0321, 2018-06-22:
@Sindorman Are you mean that edite the camera.launch file? how to modify the packet resend argument? thank you!