ros-drivers/pointgrey_camera_driver #159: How do i use this driver for Blackfly S point grey camera?

sumitspattan, 2018-02-08:
Hello,

How do i use this driver for blackfly s model?

lab135-ncepu, 2018-04-05:
i have the same problem.

sumitspattan, 2018-04-09:
@fabrizioschiano  Since the driver which is currently implemented is using FlyCapture SDK, is it possible to implement the same driver for Spinnaker SDK? 

Or

is it possible to use the existing driver for Blackfly S camera, Model: BFS-PGE-50S5C ? 

If I download the existing ros driver, it won't work. As said by  @jzx-gooner

nflsalex, 2018-05-21:
Same here, how do we make it work for Blackfly S?

DrCurry, 2018-07-24:
BlackFly S only supports Spinnaker. This driver uses FlyCapture. If you want to use BlackFly S, you are going to have to write your own node wrapping the Spinnaker SDK with ROS

mhosmar-cpr, 2018-07-24:
Hi all, I have been working on a Spinnaker driver which I have tested working with Blackfly S (BFS-U3-32S4M) and Chameleon 3.  Check it out over at https://github.com/ros-drivers/flir_camera_driver. It is based on pointgrey_camera_driver and should look fairly familiar.  If there are any issues specific to your camera I encourage you to contribute fixes. I hope that eventually we will be able to support all the new FLIR cameras.

closed 2020-04-04 by tonybaltovski