ros-drivers/pointgrey_camera_driver #162: Issue with setFormat7 when running Chameleon3

sspatank, 2018-03-07:
When I try to launch bumblebee.launch, I get the following error:

`[ERROR] [1520389180.149167419]: Reconfigure Callback failed with error: PointGreyCamera::setFormat7 Format 7 mode not supported on this camera.
`

I am trying to run Chameleon 3 on Ubuntu 16.04 i7 computer. It does get listed properly as shown below:

`Number of cameras found: 1
[0]Serial: 17453955, Model: Chameleon3 CM3-U3-31S4C, Vendor: Point Grey Research, Sensor: Sony IMX265 (1/1.8" Color CMOS), Resolution: 2048x1536, Color: true, Firmware Version: 1.12.3.0
`

fabrizioschiano, 2018-03-07:
Hi, 
it seems similar to an issue I had some time ago and I solved with this: 
https://github.com/ros-drivers/pointgrey_camera_driver/commit/db22db0c7e08230066f37f2cbd37e6012476eede

Could you check that you have the latest version of the driver?

sspatank, 2018-03-07:
Do you mean the ros package? It is the latest package.
`0.13.4-0xenial-20180115-020148-0800`
Also, the camera does show up on FlyCapture2 and I can the streaming image. 

The lines of code that are giving trouble are shown below.

` Format7Info fmt7Info;
  bool supported;
  fmt7Info.mode = fmt7Mode;
  error = cam_.GetFormat7Info(&fmt7Info, &supported);
  PointGreyCamera::handleError("PointGreyCamera::setFormat7 Could not get Format 7 information", error);
  if(!supported)
  {
    throw std::runtime_error("PointGreyCamera::setFormat7 Format 7 mode not supported on this camera.");
}`

sspatank, 2018-03-08:
I looked at your issue and you were trying to change the mode. I am just trying to get it start talking with ROS and publish some basic topics. 

I have traced the function back to Camera.h in flycapture/include/ but it is declared as a virtual function there and never overwritten anywhere else as far as I can tell.  Are there common mistakes that may prevent the camera from supporting Format7? I already increased the USB data limit.

sspatank, 2018-03-12:
I was running the Bumblebee package for my Chameleon3. I used the camera launch file and it worked.

closed 2018-03-12 by sspatank

fabrizioschiano, 2018-03-20:
Nice. Thanks for reporting it here !