ros-drivers/pointgrey_camera_driver #171: [ERROR] [1531280369.072312977]: PointGreyCamera::connect could not set GigE packet_size | FlyCapture2::ErrorType 29 Failed to write SCPS0 register

ccs-ros, 2018-07-11:
my camera is  [pointgrey GRASSHOPPER2 GS2-GE-20S4C-C]
when i use the command "roslaunch pointgrey_camera_driver bumblebee.launch",there will be a error message "[ERROR] [1531280369.072312977]: PointGreyCamera::connect could not set GigE packet_size | FlyCapture2::ErrorType 29 Failed to write SCPS0 register"

And when i use the "roslaunch pointgrey_camera_driver camera.launch",the error message is:
[FATAL] [1531280524.672868980]: Failed to load nodelet '/camera/camera_nodelet` of type `pointgrey_camera_driver/PointGreyCameraNodelet` to manager `camera_nodelet_manager'
[ERROR] [1531280524.755641718]: Failed to find nodelet with name '/camera/camera_nodelet' to unload.
[camera/camera_nodelet-2] process has died [pid 7489, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load pointgrey_camera_driver/PointGreyCameraNodelet camera_nodelet_manager __name:=camera_nodelet __log:=/home/ccs/.ros/log/269df168-84b9-11e8-9e09-b06ebf33f89e/camera-camera_nodelet-2.log].
log file: /home/ccs/.ros/log/269df168-84b9-11e8-9e09-b06ebf33f89e/camera-camera_nodelet-2*.log
[camera/image_proc_debayer-3] process has finished cleanly
log file: /home/ccs/.ros/log/269df168-84b9-11e8-9e09-b06ebf33f89e/camera-image_proc_debayer-3*.log
please give me some solutions,extremely grateful！

XHwind, 2019-04-23:
Have the same problem!

codejoey, 2020-06-26:
Same problem, perhaps the spinnaker driver could work?