ros-drivers/pointgrey_camera_driver #174: pointgrey_camera_driver launch file for synchronized mode with two mono Chameleon3 Cameras

astronaut971, 2018-07-18:
Hi Im using Ubuntu 16.04 and ROS kinetic. Would like to do stereo calibration with two mono Chamelon3 PointGrey Cameras. I found that ROS drivers for Chameleon3 pointgrey_camera_driver support only FlyCapture2 SDK, But there the GUI doesn’t work good. Is there any launch file where can set up both Chameleon3 Cameras for synchronized mode and save multiply images or video in order to calibrate both cameras using ROS stereo calibration tutorial?

astronaut971, 2018-07-18:
I used this launch file
```
<launch>
  <arg name="left_camera_serial" default="18081067" />
  <arg name="left_camera_calibrated" default="0" />
  <arg name="right_camera_serial" default="17496474" />
  <arg name="right_camera_calibrated" default="0" />
  <arg name="frame_rate" default="15" />

<group ns="left_camera">
    <node pkg="nodelet" type="nodelet" name="left_camera_nodelet_manager" args="manager" />

    <node pkg="nodelet" type="nodelet" name="left_camera_nodelet"
          args="load pointgrey_camera_driver/PointGreyCameraNodelet left_camera_nodelet_manager" >
      <param name="frame_id" value="left_camera" />
      <param name="serial" value="$(arg left_camera_serial)" />

      <!-- When unspecified, the driver will use the default framerate as given by the
           camera itself. Use this parameter to override that value for cameras capable of
           other framerates. -->
      <param name="frame_rate" value="$(arg frame_rate)" />

      <!-- Use the camera_calibration package to create this file -->
      <param name="camera_info_url" if="$(arg left_camera_calibrated)"
             value="file://$(env HOME)/.ros/camera_info/$(arg left_camera_serial).yaml" />
    </node>

    <node pkg="nodelet" type="nodelet" name="image_proc_debayer_left"
        args="load image_proc/debayer left_camera_nodelet_manager">
    </node>
  </group>

  <group ns="right_camera">
    <node pkg="nodelet" type="nodelet" name="right_camera_nodelet_manager" args="manager" />

    <node pkg="nodelet" type="nodelet" name="right_camera_nodelet"
          args="load pointgrey_camera_driver/PointGreyCameraNodelet right_camera_nodelet_manager" >
      <param name="frame_id" value="right_camera" />
      <param name="serial" value="$(arg right_camera_serial)" />

      <!-- When unspecified, the driver will use the default framerate as given by the
           camera itself. Use this parameter to override that value for cameras capable of
           other framerates. -->
      <param name="frame_rate" value="$(arg frame_rate)" />

      <!-- Use the camera_calibration package to create this file -->
      <param name="camera_info_url" if="$(arg right_camera_calibrated)"
             value="file://$(env HOME)/.ros/camera_info/$(arg right_camera_serial).yaml" />
    </node>

    <node pkg="nodelet" type="nodelet" name="image_proc_debayer_right"
        args="load image_proc/debayer right_camera_nodelet_manager">
    </node>
  </group>
</launch>

Then got this error
[ERROR] [1531907273.065364517]: Reconfigure Callback failed with error: PointGreyCamera::setExternalStrobe Could not set strobe control. | FlyCapture2::ErrorType 23 Failed to set GPIO pin direction. Pin may be optoisolated.

Any help?

closed 2018-07-18 by astronaut971

kanishkaganguly, 2021-02-26:
Any solution to this issue?
I am facing the same problem with a single camera.