ros-drivers/pointgrey_camera_driver #177: Does this driver support Hardware Synchronization with two cameras?

astronaut971, 2018-07-23:
Hi 

Im using Ubuntu 16.04 ROS kinetic and two Chameleon3 mono cameras. Im able to connect the two cameras  with GPIO cables. then also able to use FlyCap2 GUI tools to setup "Hardware Synchronization" and it works well. But when try to use the launch file im not able to make them both streaming in same time. Using rqt_image_view  I can see only the master streaming , but topic is published from both cameras. So i would like to know if this driver support Hardware Synchronization with two cameras?

oaix, 2018-08-03:
*Yes.*  
Set both cameras to `trigger mode`.

astronaut971, 2018-08-03:
You mean software wise or? Cameras are connected with their GPIO pins according FlyCapture 2 documentation to work as Master Slave in synchronized mode. So what should be done to set both cameras to trigger mode? with FlyCapture2 GUI or with software code?

astronaut971, 2018-08-06:
ok. so where to find example of this launch file?

jasonbeach, 2019-02-12:
I'm interested in this as well.  I have two Flea3's (USB3) with the trigger mode set as it should and both triggered via an autopilot gpio at 20 Hz. It works, and I can get data from both cameras, however the timestamps for the same image from each camera is different (they're close, but generally anywhere from 500 us to 1-2 ms different.  This prevents it from working with the TimeSynchronizer class.  I know I can use the ApproximateTimeSynchronizer, but it seems that I shouldn't have to.  For parameters I fixed the exposure time to 10ms to see if that would get them closer, but that doesn't seem to have made a difference.

matheusbg8, 2020-04-08:
Hi @oaix , @jasonbeach 

I am trying to synchronize 2 Flae3 GigE. I am using the trigger, and it is working. I can validate it by changing the FPS of camera1 and observing the same behavior on camera2.

However, I am having the same issues with the msg header timestamp. Some algorithms need identical timestamps, and we will never get it using two independent nodes. If you are using message_filters::TimeSynchronizer, it will never fire the callback function.

Does someone have a solution for this? 

I was looking at the code of bumblebee, and the main difference is that only one node is running and getting both images. It writes exactly the same timestamp in both msgs. So it will work on stereo vision algorithms.