ros-drivers/pointgrey_camera_driver #179: Failed to load nodelet '/camera/camera_nodelet` of type `pointgrey_camera_driver/PointGreyCameraNodelet` to manager `camera_nodelet_manager

campergat, 2018-09-20:
Hello,
I have not been able to launch the driver to work with my Blackfly camera. (I am running ubuntu 16.04 and ros kinetic on my computer).
I can run list_cameras with no issues:
Number of cameras found: 1
`[0]Serial: 18288278, Model: Blackfly BFLY-PGE-13H2C, Vendor: Point Grey Research, Sensor: Sharp RJ33J3CA0DT (1/3" Color CCD), Resolution: 1288x964, Color: true, Firmware Version: 1.62.3.0`
and I can also run the flycap software and view the image stream successfully. However, when I try to launch the driver with roslaunch pointgrey_camera_driver camera.launch, It crashes with the following errors:
```
[FATAL] [1537455295.650868657]: Failed to load nodelet '/camera/image_proc_debayer` of type `image_proc/debayer` to manager `camera_nodelet_manager'
[FATAL] [1537455295.650871057]: Failed to load nodelet '/camera/camera_nodelet` of type `pointgrey_camera_driver/PointGreyCameraNodelet` to manager `camera_nodelet_manager'
[camera/camera_nodelet_manager-2] process has died [pid 6581, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=camera_nodelet_manager __log:=/home/user/.ros/log/22b8e21a-bce5-11e8-bba4-704d7b6503a4/camera-camera_nodelet_manager-2.log].
log file: /home/user/.ros/log/22b8e21a-bce5-11e8-bba4-704d7b6503a4/camera-camera_nodelet_manager-2*.log
[camera/camera_nodelet-3] process has died [pid 6582, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load pointgrey_camera_driver/PointGreyCameraNodelet camera_nodelet_manager __name:=camera_nodelet __log:=/home/user/.ros/log/22b8e21a-bce5-11e8-bba4-704d7b6503a4/camera-camera_nodelet-3.log].
log file: /home/user/.ros/log/22b8e21a-bce5-11e8-bba4-704d7b6503a4/camera-camera_nodelet-3*.log
[camera/image_proc_debayer-4] process has died [pid 6588, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load image_proc/debayer camera_nodelet_manager __name:=image_proc_debayer __log:=/home/user/.ros/log/22b8e21a-bce5-11e8-bba4-704d7b6503a4/camera-image_proc_debayer-4.log].
```

I have also tried to run each nodelet separately from the command line in order. This results in the manager crashing with a segfault the moment I try to run pointgrey_camera_driver/PointGreyCameraNodelet.

am I missing something here? Does this error look familiar to anyone? Any help would be much appreciated.

salsicha, 2018-09-28:
same problem, is this (0.13.4-0xenial-20180824-172336-0800) package not working for anyone else?

salsicha, 2018-09-28:
I think I need to roll back to (ros-kinetic-pointgrey-camera-driver/now 0.13.4-0xenial-20180309-130427-0800 amd64). How can I do that? Are the older builds stored anywhere?

AtDinesh, 2018-11-13:
same issue for me !

stoneil, 2019-03-20:
Hi, I am having the same problem but with the Chameleon 3 Camera and ROS Kinetic. I know this is an old issue but were you able to find any fixes to this problem? I have tried installing the Fly Capture 2 library drivers for my machine and still get the same bug

campergat, 2019-03-26:
I honestly don't remember exactly what I did to fix it, but it had something to do with incompatible libraries. I believe I solved this by compiling from source with the libraries I had.

cesarhcq, 2019-07-25:
Anyone solved this issue? Thank you!

ConnorNovak, 2019-11-13:
I am also currently having this issue with ROS Kinetic and Ubuntu 16.04, Using a Pointgrey Flea.

civerachb-cpr, 2020-12-01:
2020 and I'm still seeing this problem on 18.04 with Melodic and a Blackfly camera.