ros-drivers/pointgrey_camera_driver #183: Support melodic

concavegit, 2019-01-09:
Would it be possible to support the ROS melodic release?

mikepurvis, 2019-04-25:
Are we just needing a release here? Has anyone built the driver from source and confirmed it works on Bionic?

florisvb, 2019-04-26:
I can confirm that installing from source on ubuntu 18.04.2 running melodic does work, after also installing FlyCapture 2.13.3.31 SDK - Linux Ubuntu 18.04 (64-bit) — 10/25/2018.

Some more detailed instructions that may be helpful to some...

Tested with a blackfly or flea USB3 (I can't recall how to tell which one).

FlyCapture 2.13.3.31 SDK can be downloaded from: https://www.ptgrey.com/support/downloads
(make a free account first, and follow the prompts to find your product and it will give the appropriate download link). 

It may be necessary to install some other packages first (see: https://www.ptgrey.com/tan/10548):
`user$: sudo apt-get install libraw1394-11 libgtkmm-2.4-1v5 libglademm-2.4-1v5 libgtkglextmm-x11-1.2-dev libgtkglextmm-x11-1.2 libusb-1.0-0`

Then run
`user$: sudo sh install_flycapture.sh`
where install_flycapture.sh is in the FlyCapture 2.13.3.31 SDK download. 

Also note that for USB3 to work properly, you may need to follow these instructions (configuring USB-FS):
https://www.ptgrey.com/KB/10685

To permanently modify the USB-FS memory limit:

In a text editor, open the /etc/default/grub file.
Find
`GRUB_CMDLINE_LINUX_DEFAULT="quiet splash"`
Replace with
`GRUB_CMDLINE_LINUX_DEFAULT="quiet splash usbcore.usbfs_memory_mb=1000"`
Update grub with these settings:
`$ sudo update-grub`
Reboot and test a USB camera.

furushchev, 2019-12-17:
@mikepurvis 

FLIR currently asks users to use `Spinnaker` driver because of deprecation of `flycapture` driver, but the spinnaker driver for ROS ( https://github.com/ros-drivers/flir_camera_driver ) also depends on some packages on this repository.

As far as I see, the dependencies (`wfov_camera_msgs` and `image_exposure_msgs`) only contain message definitions, so I'd like to ask maintainers to release these packages first on melodic for `spinnaker_camera_driver` users, or should we move these packages to `flir_camera_driver` repos?

gftabor, 2020-03-10:
Has there been any progress on this? Even someone's person forks that does the required rearranging?

ayuguchi, 2020-03-19:
@gftabor Hi! FYI, if you use a FLIR camera, you can use this repository.
https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver
This repo can also work on ROS melodic when I use a FLIR camera.

tonybaltovski, 2020-04-04:
Has been bloomed: https://github.com/ros/rosdistro/pull/24348

closed 2020-04-04 by tonybaltovski