ros-drivers/pointgrey_camera_driver #188: Debians missing

impala454, 2019-06-11:
We use the pre-built debian packages from this repo and noticed that since the GPG-pocalypse the debians for this repo are missing.  The debians for [ros-kinetic-pointgrey-camera-description](http://packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-pointgrey-camera-description/)  are there but [ros-kinetic-pointgrey-camera-driver](http://packages.ros.org/ros/ubuntu/pool/main/r/ros-kinetic-pointgrey-camera-driver/) debians are missing.  Any help would be appreciated.

srsidd, 2019-06-11:
This is related to https://github.com/ros-drivers/pointgrey_camera_driver/issues/187 and https://discourse.ros.org/t/new-gpg-keys-deployed-for-packages-ros-org/9454. Since the ros GPG keys uploaded, the pointgrey binaries are now held at the new `packaged.ros.org`. As per #187, the builds no longer work. So we will not have debians till the builds work, and the binary gets uploaded to the new server.

Durant35, 2019-07-26:
When will it fix?

dylancharter, 2019-08-15:
If you don't want to compile the driver from source, I was able to get it from the snapshot repository: http://wiki.ros.org/SnapshotRepository. I didn't add the whole source list, but just downloaded the .deb [http://snapshots.ros.org/kinetic/2019-05-10/ubuntu/pool/main/r/ros-kinetic-pointgrey-camera-driver/](here) for kinetic and ran the following to install.

`sudo dpkg -i /the/absolute/path/to/deb/file
sudo apt-get install -f`

I could see this method causing dependency issues so it may be better to use what's in the ros wiki, but this worked for me.

tonybaltovski, 2020-04-04:
Deb is currently in shadow-fixed.

closed 2020-04-04 by tonybaltovski