ros-drivers/pointgrey_camera_driver #189: Cannot Download Camera Driver

thefinaljob, 2019-06-28:
Hello There,

As other users have pointed out, we are unable to successfully install the driver using sudo. We are getting the following error.

`Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-lunar-pointgrey-camera-driver`

I was wondering if it would be possible for this issue to be rectified.

hcyoo93, 2019-10-25:
I have same problem
How did you resolve this issue?

anakin0620, 2019-12-03:
https://github.com/ros-drivers/pointgrey_camera_driver/issues/188#issuecomment-521527548

tonybaltovski, 2020-04-04:
Has been released for Kinetic and Melodic.  Lunar is EOL.

closed 2020-04-04 by tonybaltovski

LukeAI, 2020-04-04:
@tonybaltovski 

The same issue applies to Kinetic. The [wiki ](http://wiki.ros.org/pointgrey_camera_driver) advises: "sudo apt-get install ros-kinetic-pointgrey-camera-driver"

but no such package exists.

tonybaltovski, 2020-04-07:
It hasn't been synced to [main](http://repositories.ros.org/status_page/kinetic_default.html) yet, you can add the [ros-shadow-fixed](http://wiki.ros.org/action/show/TestingRepository?action=show&redirect=ShadowRepository) if you want the deb.