ros-drivers/pointgrey_camera_driver #197: Rename statistics_msgs?

gavanderhoorn, 2020-04-22:
Though the package has existed for a *long* time (since f9ef7cd it seems), the name `statistics_msgs` seems like it's really too generic.

I just had a clash with an equally badly named package (mea culpa), but I'm at least not releasing the result of my sin to the wider ROS community ;)

Has there ever been any discussion about renaming `statistics_pkg` to something more REP-144 compliant?

mikepurvis, 2020-04-22:
Yeah, I'm the guilty party here. Per the copyright notice, this package was originally an internal affair at CMU, and I'm the one who figured out the hacks to distribute flycapture and get it shipped. It was a rush to get it out the door, so there wasn't much in the way of API review at the time, even internally (at Clearpath).

@tonybaltovski also mentioned interest in migrating the messages from this repo into `sensor_msgs` or somewhere else common. Such a thing could also be a good catalyst for an API review.

gavanderhoorn, 2020-04-23:
So what would be a good way to get this started?

Would this be something to do before a Noetic release?

tonybaltovski, 2020-04-23:
Looking at `statistics_msgs`, there is only 1 message that is used by `image_exposure_msgs`.  That one message could just be moved into `image_exposure_msgs`.  Moving all the messages to a separate repo in this step would make sense as well.  This could definitely be done before Noeitc is released.  Looking at the messages here, I worry they won't be generic enough for `sensor_msgs`.

gavanderhoorn, 2020-04-23:
Seeing as I'm the one complaining: would you want me to submit a PR?

tonybaltovski, 2020-05-05:
Sure, I can make a noetic branch.  Moving it into `image_exposure_msgs` should be pretty contained.  Simialry, moving the messages to a seprate repo will be no change.