ros-drivers/pointgrey_camera_driver #202: Blackfly BFS-U3-04S2C-CS - No Cameras Detected

civerachb-cpr, 2020-09-23:
I'm working on a robot with 2x Blackfly BFS-U3-04S2C-CS cameras connected to it, but the driver is not detecting either of them.

Running `rosrun pointgrey_camera_driver list_cameras` just prints:

No PointGrey cameras detected on this computer.

Note that you may need to restart udev and replug your camera, eg:
  sudo service udev restart

I've modified the pointgrey udev rules to add the attributes for this camera:
`ATTR{idVendor}=="1e10", ATTR{idProduct}=="4000", MODE="0666", GROUP="pgrimaging"`
based on the output of udevadm info:

$udevadm info --attribute-walk /dev/bus/usb/004/002

  looking at device '/devices/pci0000:00/0000:00:14.0/usb4/4-1':
    KERNEL=="4-1"
    SUBSYSTEM=="usb"
    DRIVER=="usb"
    ATTR{authorized}=="1"
    ATTR{avoid_reset_quirk}=="0"
    ATTR{bConfigurationValue}=="1"
    ATTR{bDeviceClass}=="ef"
    ATTR{bDeviceProtocol}=="01"
    ATTR{bDeviceSubClass}=="02"
    ATTR{bMaxPacketSize0}=="9"
    ATTR{bMaxPower}=="896mA"
    ATTR{bNumConfigurations}=="1"
    ATTR{bNumInterfaces}==" 3"
    ATTR{bcdDevice}=="0000"
    ATTR{bmAttributes}=="80"
    ATTR{busnum}=="4"
    ATTR{configuration}==""
    ATTR{devnum}=="2"
    ATTR{devpath}=="1"
    ATTR{idProduct}=="4000"
    ATTR{idVendor}=="1e10"
    ATTR{ltm_capable}=="no"
    ATTR{manufacturer}=="FLIR"
    ATTR{maxchild}=="0"
    ATTR{product}=="Blackfly S BFS-U3-04S2C"
    ATTR{quirks}=="0x0"
    ATTR{removable}=="removable"
    ATTR{serial}=="0136489B"
    ATTR{speed}=="5000"
    ATTR{urbnum}=="15"
    ATTR{version}==" 3.10"

  looking at parent device '/devices/pci0000:00/0000:00:14.0/usb4':
    KERNELS=="usb4"
    SUBSYSTEMS=="usb"
    DRIVERS=="usb"
    ATTRS{authorized}=="1"
    ATTRS{authorized_default}=="1"
    ATTRS{avoid_reset_quirk}=="0"
    ATTRS{bConfigurationValue}=="1"
    ATTRS{bDeviceClass}=="09"
    ATTRS{bDeviceProtocol}=="03"
    ATTRS{bDeviceSubClass}=="00"
    ATTRS{bMaxPacketSize0}=="9"
    ATTRS{bMaxPower}=="0mA"
    ATTRS{bNumConfigurations}=="1"
    ATTRS{bNumInterfaces}==" 1"
    ATTRS{bcdDevice}=="0415"
    ATTRS{bmAttributes}=="e0"
    ATTRS{busnum}=="4"
    ATTRS{configuration}==""
    ATTRS{devnum}=="1"
    ATTRS{devpath}=="0"
    ATTRS{idProduct}=="0003"
    ATTRS{idVendor}=="1d6b"
    ATTRS{interface_authorized_default}=="1"
    ATTRS{ltm_capable}=="yes"
    ATTRS{manufacturer}=="Linux 4.15.0-115-generic xhci-hcd"
    ATTRS{maxchild}=="6"
    ATTRS{product}=="xHCI Host Controller"
    ATTRS{quirks}=="0x0"
    ATTRS{removable}=="unknown"
    ATTRS{serial}=="0000:00:14.0"
    ATTRS{speed}=="5000"
    ATTRS{urbnum}=="58"
    ATTRS{version}==" 3.00"

  looking at parent device '/devices/pci0000:00/0000:00:14.0':
    KERNELS=="0000:00:14.0"
    SUBSYSTEMS=="pci"
    DRIVERS=="xhci_hcd"
    ATTRS{broken_parity_status}=="0"
    ATTRS{class}=="0x0c0330"
    ATTRS{consistent_dma_mask_bits}=="64"
    ATTRS{d3cold_allowed}=="1"
    ATTRS{device}=="0x8c31"
    ATTRS{dma_mask_bits}=="64"
    ATTRS{driver_override}=="(null)"
    ATTRS{enable}=="1"
    ATTRS{irq}=="29"
    ATTRS{local_cpulist}=="0-3"
    ATTRS{local_cpus}=="f"
    ATTRS{msi_bus}=="1"
    ATTRS{numa_node}=="-1"
    ATTRS{revision}=="0x05"
    ATTRS{subsystem_device}=="0x8c31"
    ATTRS{subsystem_vendor}=="0x8086"
    ATTRS{vendor}=="0x8086"

  looking at parent device '/devices/pci0000:00':
    KERNELS=="pci0000:00"
    SUBSYSTEMS==""
    DRIVERS==""

Adding a SYMLINK+=[...] shows that the rule is being triggered.

Possibly related (possibly not?) when I try to run the driver, specifying the serial numbers for the left & right cameras shows that the serial numbers are being truncated:

[ERROR] [1600872764.137690069]: Reconfigure Callback failed with error: PointGreyCamera::connect Could not find camera with serial number: 1364217. Is that camera plugged in? | FlyCapture2::ErrorType 19 Failed to get the camera from the serial number.
[ERROR] [1600872764.449145360]: Reconfigure Callback failed with error: PointGreyCamera::connect Could not find camera with serial number: 136489. Is that camera plugged in? | FlyCapture2::ErrorType 19 Failed to get the camera from the serial number.

but in my launch file I have:

  <arg name="left_camera_serial" default="0136489B" />
  <arg name="left_camera_calibrated" default="0" />

  <arg name="right_camera_serial" default="01364217" />
  <arg name="right_camera_calibrated" default="0" />

with those args being passed to the camera nodelets' serial parameters.  The leading 0s are being stripped off, as is the trailing B on the left camera's serial number.  I'm not too worried about the leading 0s, but the trailing B seems important.

I'm not sure how to proceed from here.  Any guidance or suggestions would be appreciated.

tonybaltovski, 2020-09-23:
@civerachb-cpr those are Blackfly S cameras, they need to use the `flir_camera_driver`.

civerachb-cpr, 2020-09-23:
Ah, good to know!

closed 2020-09-23 by civerachb-cpr

Hateley123, 2022-09-02:
I am getting this error: 
No PointGrey cameras detected on this computer.

Note that you may need to restart udev and replug your camera, eg:
  sudo service udev restart

I have checked the udev rules and they are correct, and I have downloaded the flir_camera_driver, but I am still getting the same error. is there anythng additionalo do to make it work?