ros-drivers/pointgrey_camera_driver #208: Compressed image transport support

ingramator, 2021-06-28:
Hi All,

Currently this driver doesn't appear to support [compressed_image_transport](http://wiki.ros.org/compressed_image_transport) which means recording the image topic will use an absurd amount of disk space. This also affects streaming for viewing purposes (like over a wireless connection).

Are there any plans to implement such support or does anyone know of any forks that do?

Cheers

mobiiin, 2022-04-16:
> I'm not sure if this helps but you can republish the stream ([compressed_image_transport, Section 4](http://wiki.ros.org/compressed_image_transport)) and get the compressed stream that way
> 
> ```
> rosrun image_transport republish raw in:=camera/image_raw out:=camera/image_repub
> ```
> 
> It is a bit of a hack solution but is what I've been using.

do you mean doing sth like this?
`rosrun image_transport republish raw in:=/camera/color/image_raw/compressed  out:=camera/image_repub image_transport:=compressed`
if so the realsense complains about the encoding needed by the subscriber:
`[ERROR] [1650095024.946461092]: Client [/image_republisher_1650095024664204994] wants topic /camera/color/image_raw/compressed to have datatype/md5sum [sensor_msgs/Image/060021388200f6f0f447d0fcd9c64743], but our version has [sensor_msgs/CompressedImage/8f7a12909da2c9d3332d540a0977563f]. Dropping connection.
`
could you help me out?

Vergilwwj, 2022-04-16:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。